mightymos/RF-Bridge-OB38S003 #4: How to flash ?

the-mentor, 2024-01-10:
Hi, 
If i want to test this firmware what is the process of flashing it to the chip?
Is it possible to use a chip clamp to flash the chip or de-soldering and re-soldering it the only way?

Thank you very much for your work !!

-DM

mightymos, 2024-01-11:
You can try to use this sketch with library:
https://github.com/mightymos/OnbrightFlasher

The sketch is intended to be run on the built in ESP8265.
But you'll have to use an external Arduino board.
This is because the RESET pin functionality is probably disabled on Sonoff (set to GPIO) but handshake needs to occur at microcontroller startup.
If you power the whole board up both microcontroller and ESP8265 will start up but ESP8265 startup is slower and so handshake will miss the microcontroller.

Instead you can have the separate Arduino powered, and have I2C pins connected to the Sonoff microcontroller.
The Arduino will be constantly trying to handshake while powered.
Then apply power to the Sonoff and handshake should occur.

If it does not complete handshake, will need to power cycle Sonoff until it does.

Then you'll need to erase chip to unprotect it.
Then can finally write hex file line by line to microcontroller.

The sketch does not easily set configuration bytes now.
So that will need to be added eventually to make this whole process easier (to use built in ESP8265 to flash).
Also some script on PC side will be needed to make sending hex file lines easier.

Do you have an Arduino board around, maybe something like an ESP32?

I can help with hardware hookup but need to know what you have first.
Possibly will try on my side first too to make sure it can work.

the-mentor, 2024-01-11:
@mightymos thanks for reaching out.
I have a bunch of ESP32 and D1 mini(esp8266) boards around. 

A step by step guide + a picture of the pins on how the wiring to the ESP32 board etc need to be wired will help a lot.

mightymos, 2024-01-13:
I have been able to erase the microcontroller with an external ESP32 board.
I am working on setting the configuration bytes so the built in ESP8265 can be used afterward because I think that will be easier.
I will respond once I have made progress.

the-mentor, 2024-01-14:
@mightymos can you write the instructions to erase the micro-controller with an external ESP32 board so I can test if I can get that working?

mightymos, 2024-01-14:
You can follow the usage instructions here now:
https://github.com/mightymos/OnbrightFlasher/blob/main/usage.txt

However, if something is unclear please ask me questions.

It is best to power both boards up from separate 3.3V supplies.
Do you have a couple usb to serial adapter boards with 3.3V output?

[image]

the-mentor, 2024-01-14:
@mightymos can you explain how I'm suppose to connect the pins ?
the picture doesnt really explain it
from which pins on the d1 to which pins on the rf bridge ?

mightymos, 2024-01-14:
You need to install the SoftWire library in the Arduino IDE.
You might also need to install esp8266 board support:
https://randomnerdtutorials.com/how-to-install-esp8266-board-arduino-ide/

I am assuming this matches your D1 board, let me know if it does not:
https://randomnerdtutorials.com/esp8266-pinout-reference-gpios/

So the D1 board is, I assume, connected to computer by USB to upload.
Then GPIO4 SDA (pin D2) on D1 board is connected to SDA pin on Sonoff (J3 connector).
Also GPIO4 SCL (pin D1) on D1 board is connected to SCL pin on Sonoff.

On Sonoff, you can have ground connected to your separate power supply.
Once you are ready to try handshake, you can apply 3.3V to Sonoff 3.3V pin (any one should work, but might make sense to just use J3 connector).

the-mentor, 2024-01-14:
@mightymos did this work properly ? 

[image]

mightymos, 2024-01-14:
It looks like the commands might have worked, except the chip read should be 'A' for this microcontroller.
You will have to redo handshake with powerup and see if it displays 'A'.

Also, you did not write any hex to microcontroller?
You can try sending all lines from blink.ihx once you see chip read 'A'.

the-mentor, 2024-01-15:
ok so after a bunch of fiddling I managed to flash it and get the red led to blink. 
now what ?

mightymos, 2024-01-15:
That's great.

The  project hex file is too big for this type of manual flashing right now.
Because it includes a much of other stuff besides pass through mode.

Let me create a simpler version and we can try to flash that.

the-mentor, 2024-01-15:
I wonder if we can modify the flasher program to flash the hex automatically rather then pasting line by line.
It can help a lot

mightymos, 2024-01-15:
That is true.

I see two ways:
[1] Send file line by line with a script on computer side
[2] Modify or replace sketch with ability to handle an entire file sent over serial port

Either approach would be helpful long term but I could not do either quickly.

I will try to make a simpler version of the pass through mode first.

mightymos, 2024-01-15:
I created a branch that tracks the simpler pass through mode firmware.
Hopefully in the future the source code can be better organized.
And/or the flashing can be made easier.

In either case, I have attached a simple pass through mode hex.
I have also attached my ESPHome yaml which demonstrated pass through is working.

[RF-Bridge-OB38S003_SimplePassThrough_01-14-2024.zip](https://github.com/mightymos/RF-Bridge-OB38S003/files/13933856/RF-Bridge-OB38S003_SimplePassThrough_01-14-2024.zip)

[sonoff_bridge.zip](https://github.com/mightymos/RF-Bridge-OB38S003/files/13933861/sonoff_bridge.zip)

Are you using ESPHome?

the-mentor, 2024-01-16:
I'm going to try and flash it sometime today and report back.

I have two rf bridges and the one I used to try to flash the rf chip's firmware has Tasmota but I'm going to flash it with esphome since that is my preferred custom firmware.

Either way thank you very much for all the support you've given me.

Once I get everything working I'll try to write documentation on how to flash the firmware and contribute it to the repo.

the-mentor, 2024-01-16:
@mightymos  I was able to flash the code and get ESPHome going.

Does pass-through mode also works for transmitting or only receiving ? 

I added this code to my ESPHome config and It doesnt turn on the device I want it to turn on.

```
switch:
  - platform: template
    name: RF Power Button
    optimistic: true
    turn_on_action:
      - remote_transmitter.transmit_rc_switch_raw:
          code: '10111100110001100101010110101'
          protocol: 1
          repeat:
            times: 1
            wait_time: 0s   
```

mightymos, 2024-01-17:
Are you using Home Assistant to toggle switch?
If so you might want to remove device from Home Assistant and re-add.
I did change the switch name to RF Transmit, so in my case RF Power would no longer exist.

I have done transmission from OB38S003 previously but it was not very thorough.
The firmware when in serial mode does not really yet support transmission.
I am not sure about other suggestions.

We can keep trying to exchange ideas until it works.

mightymos, 2024-01-17:
I did want to comment that my receiver with Tasmota is a Sonoff R2 v1.0 (black box) running Portisch firmware.

the-mentor, 2024-01-17:
If I understand you correctly the firmware  in serial mode (pass through) doesn't support transmitting yet.

I think it will be great to add that feature.
Once that feature is added this firmware should be as good as the direct hack just without the hard work of soldering/cutting traces etc.

Either way amazing work!!

mightymos, 2024-01-18:
Do you have an oscilloscope or logic analyzer?

the-mentor, 2024-01-18:
> Do you have an oscilloscope or logic analyzer?

I do not.

mightymos, 2024-01-18:
You might be able to use something like this to inspect the transmit pin with another of your D1 boards:
https://github.com/aster94/logic-analyzer

There are also logic analyzers that work with pulseview for about $20 USD.

That would at least indicate if the transmit pin is toggling or not.

the-mentor, 2024-01-18:
What is the transmit pin on the j3 connector that I need to monitor?

the-mentor, 2024-01-18:
Nevermind about the sending issue I was able to confirm that sending works.

mightymos, 2024-01-18:
> Nevermind about the sending issue I was able to confirm that sending works.

That's great, was it something simple, how were you able to confirm?

the-mentor, 2024-01-18:
I'm tinkering with all kinds of RF devices since I'm trying to control my Rf fans.

So I got all kinds of devices from aliexpress in order to try and reverse engineer the remote.

One of them is the cc1101 transponder that arrived a few days ago and I was able to set it up today and I could see the rf signals coming in from the sonoff rf bridge with the serial pass through firmware.

Your excellent work makes it very easy to modify the device without too much Hassel and no soldering requirments.

mightymos, 2024-01-18:
For future reference here are pin descriptions that can be further notated:
https://github.com/mightymos/RF-Bridge-OB38S003/blob/main/notes/sonoff_r2v2.2_connector_pinouts_fv1.txt

That's great you saw some radio activity.
Keep testing and do let me know how things go.

syepes, 2024-01-27:
@mightymos
First of all thanks for the hard work on this!
I don't have a spear ESP32 board for flashing the MCU, so I am going to buy a [SyncMOS MSM9066 MCU Programmer MCU](https://www.ebay.com/itm/354807037482). Could you share the procedure and pinout when using this method.

the-mentor, 2024-01-28:
@syepes i highly recommend you buy a nodemcu or a esp8266 d1 mini
They are much cheaper and easier to flash
Also I recommend you get a serial flasher that has 3.3v output or a 3.3v battery to power the chip for during flashing.
Also header pins helped a lot during flashing and they prevent you from needing to solder anything

syepes, 2024-01-28:
Thanks I am receiving a esp8266 d1 mini this week and I already have the series flasher.
It would be appreciated the procedure on what need to be flashed on the D1 and than where are the last hex lines that I need to flash.

I also have two Sonoff R2 one with the direct hack + esphome and one unmodified that I'll be testing with this method

mightymos, 2024-01-28:
You'll need to upload the OnbrightFlasher sketch to the ESP8266 D1 mini or similar board.
You could actually have used one of the Sonoff ESP8265 to flash the microcontroller on another board.
But having the separate mini is probably convenient.

You can go to the Releases page of this repository to get the PassthroughMode hex file for the microcontroller.
There are instructions in the readme and a usage.txt at the OnbrightFlasher repository.
They probably should be combined and edited.
If something is not clear just ask for help.

If you did for some reason want to use the official MSM9066 programmer, here is a picture of pins:
[image]

mightymos, 2024-01-28:
Here are the SMAP AC settings for the MSM9066 official programmer:
[image]

syepes, 2024-01-30:
@mightymos

I have finally gone with the ESP8266 D1 board, but I am seeing these error when powering the Sonoff.

[image]

**This is the pin connection:**
- GPIO4 SCL (pin D1) on D1 board is connected to SCL pin on Sonoff
- GPIO4 SDA (pin D2) on D1 board is connected to SDA pin on Sonoff

Do you know what could be wrong?

mightymos, 2024-01-30:
The software I2C library is probably taking too long to time out (if nothing is connected to bus) and then the software watchdog is thinking the program is stuck so it forces a reset. I probably need to improve the sketch so this does not happen.

I think you can make it work.
Have ground connected on the target but 3.3V disconnected on target.

Upload the sketch to the D1 board.
Then immediately change to serial monitor and type "handshake".
Then apply 3.3V to target board.

I think target should then detect handshake and other instructions can be followed.

syepes, 2024-01-30:
Thanks for the pointers, but I have been trying for 1h and could not get it to display `Connected...`
I can not figure out whats wrong, I have double checked the pins and they look correct, I have also tried different timings when applying the power but no luck :-(

[image]

mightymos, 2024-01-31:
It looks correct...it is always a possibility that without soldering in headers you have a loose connection(s).
I think you should try it by leaving the 3.3 V Power Supply plugged into USB but leave the VCC pin on the target end disconnected.
Then when you are ready to apply power touch and leave 3.3V pin touching.

If you cannot get it to work we can investigate further.

mightymos, 2024-01-31:
What output are you seeing in serial monitor?

mightymos, 2024-01-31:
I pushed changes to the OnbrightFlasher repo.
I shortened the softwire i2c timeout so that the watchdog on the processor should no longer reset while waiting in handshake state.

I get this output in the serial monitor now:
```
Ready.
Entering [idle] state.
Type [handshake] to attempt connection to target.
Type [idle] and then [handshake] to start from the beginning
handshake
Reset to handshake

Chip failed to read
Can try command [signature] over serial port to see if Chip read: A
signature
Read chip type...Chip read: A
```

syepes, 2024-01-31:
Thanks for the update, I am still having the same issue. I never see the `Connected...` string message in the monitor. I have also tried another D1 board :-(
Question, when powering the Bridge on VCC do we need to press the reset button or something else that I am missing?

mightymos, 2024-01-31:
You should not need to hold the reset button on the bridge during power up.
The reset button is only used to place the ESP8265 in boot mode.

With the fixes now, you should just need to upload sketch to D1 board once.
Just leave it running and can issue commands.
Type [idle] to start over, followed by [handshake] if retrying handshake.
Then apply power to target bridge.

As you can see in my output above, for some reason now the chip read is failing during handshake.
However handshake must have worked because I am able to issue [signature] command and read a hex "A" which is correct for the chip type.

What serial output do you get when you try?

mightymos, 2024-02-02:
I've updated OnbrightFlasher and the handshake seems to be reliable now.
Appreciate anyone that can test on D1 board.

matlab22, 2024-02-02:
I would like to test it but i just have a esp32 s3 mini

the-mentor, 2024-02-02:
> I've updated OnbrightFlasher and the handshake seems to be reliable now.
> Appreciate anyone that can test on D1 board.

I'll try to so some testing tonight

mightymos, 2024-02-02:
@matlab22 I added support for ESP32 which works on my ESP-WROOM-32. But I need to know exactly what gpio pins are available on your board. Can you provide a link to the actual board you purchased?

@syepes I realized the instructions were unclear in readme to say the least...one potential problem is in Arduino IDE you need to set serial monitor to "Both NL & CR" at 115200 baud. Otherwise the command parsing will never work. I have gotten ESP8265 to sonoff target and ESP32 to sonoff target working reliably and others are testing.

@the-mentor Thanks, appreciate the help for syepes.

matlab22, 2024-02-02:
I have a yd-esp32-23 hosting an esp32-s3-wroom-1.
https://mischianti.org/vcc-gnd-studio-yd-esp32-s3-devkitc-1-clone-high-resolution-pinout-and-specs/

matlab22, 2024-02-02:
I just use the esp32s3.name=ESP32S3 Dev Module

mightymos, 2024-02-02:
> I just use the esp32s3.name=ESP32S3 Dev Module

Ok, I added support for the ESP32S3 and arbitrarily used gpio4 and gpio5 for SDA and SCL pins.

matlab22, 2024-02-02:
> > I just use the esp32s3.name=ESP32S3 Dev Module
> 
> Ok, I added support for the ESP32S3 and arbitrarily used gpio4 and gpio5 for SDA and SCL pins.

Perfect thank you 👍 I guess then I have to pass the 23 lines from https://github.com/mightymos/RF-Bridge-OB38S003/releases/download/v0.2.0/RF-Bridge-OB38S003_PassthroughMode.hex
the 192 from the serial mode are a bit to much. Doing it by hand. I might make a python script handling it or do you have already something?

mightymos, 2024-02-02:
> Perfect thank you 👍 I guess then I have to pass the 23 lines from https://github.com/mightymos/RF-Bridge-OB38S003/releases/download/v0.2.0/RF-Bridge-OB38S003_PassthroughMode.hex the 192 from the [...]

I do not know any python so have no script unfortunately.
You are right, manual copying is tiresome.
Fortunately, if the passthrough flashing works the first time then it only needs to be done once.
However, for the long term a script or way to upload the entire hex would be preferred.

the-mentor, 2024-02-06:
I got around to testing the updated flasher and it seems like it was much easier this time around

mightymos, 2024-02-06:
> I got around to testing the updated flasher and it seems like it was much easier this time around

With the ESP32 module I have I seem to get write errors during hex flash. Then I have to repeat sending hex lines.
With the ESP8265 in sonoff used to flash another target sonoff I do not get errors while writing.

I am not sure what is going on with the errors on ESP32.

the-mentor, 2024-02-06:
I tried to use ChatGPT to find a way to add a command to the Arduino code to flash all the lines one after another but my C++ skills are terrible.

I also tried to write a python script that uses serial but because of all the back and forth between reading and writing I also failed to make the process easier :(

mightymos, 2024-02-06:
> I tried to use ChatGPT to find a way to add a command to the Arduino code to flash all the lines one after another but my C++ skills are terrible.
>
> I also tried to write a python script that uses [...]

Thanks for looking into it, there would be a learning curve for sure.

matlab22, 2024-02-06:
> Thanks for the pointers, but I have been trying for 1h and could not get it to display `Connected...` I can not figure out whats wrong, I have double checked the pins and they look correct, I have also tried different timings when applying the power but no luck :-(
> 
> ![CleanShot 2024-01-30 at 10  49 24@2x](https://private-user-images.githubusercontent.com/333081/300971613-76a46676-db43-4474-b53f-10e32fcaee08.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDcyNTM1MDksIm5iZiI6MTcwNzI1MzIwOSwicGF0aCI6Ii8zMzMwODEvMzAwOTcxNjEzLTc2YTQ2Njc2LWRiNDMtNDQ3NC1iNTNmLTEwZTMyZmNhZWUwOC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMjA2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDIwNlQyMTAwMDlaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1kNzFiMGRhMjA0MzVkMTdhYmRkNmMwYWE4MjMzY2I3ZGQ5ZWNkZTdiMmI3OGVkYjcwY2QyZDkwYTdlNjU5NWUzJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.QaCkAKThCS5inUvrnXypYFqm_Hms_uutDkhC_VYMcuY)
> 
> @the-mentor Did you have to do something special with the D1 flashing?

It looks like your setup is missing a common gnd.

mightymos, 2024-02-06:
> It looks like your setup is missing a common gnd.

That's a valid observation but the ground through the USB connector on both devices should I think be common. It could be tested with multimeter by attaching one lead to ground on USB connector at PC side and seeing if voltage can be measured on each device (e.g. 3.3 V).

matlab22, 2024-02-06:
I had a lot of error messages, turned out I had to read the instructions fully before i succeeded:

```
:1D01CF0002C290AF09A291E433F509EFB5090280D7E5096004D28780CFC28780CB2B
Wrote 29 bytes

:06003500E478FFF6D8FD9F
Wrote 6 bytes

:200013007900E94400601B7A009001EC780175A000E493F2A308B8000205A0D9F4DAF275A6
Wrote 32 bytes

:02003300A0FF2C
Wrote 2 bytes

:20003B007800E84400600A790175A000E4F309D8FC7800E84400600C7900900001E4F0A3C3
Wrote 32 bytes

:04005B00D8FCD9FAFA
Wrote 4 bytes

:0D00060075810912014FE5826003020003BD
Wrote 13 bytes

:00000001FF

mcureset
MCU reset...
```

I will make a python script to write the hex files and post it here.

matlab22, 2024-02-07:
make a new function reset_mcu where you send mcureset and the respons should be mcureset new line

MCU reset...

matlab22, 2024-02-07:
Here is the script I made: (https://github.com/mightymos/OnbrightFlasher/pull/1)

the-mentor, 2024-02-07:
@matlab22 very cool script.
I will test it out when I get a chance and provide feedback

the-mentor, 2024-02-07:
@matlab22 the script worked great. 
I think you can add a step to do the `setfuse 18 249` every time since it dosen't cause any issues if you do it every time.

[91 more replies]

closed 2024-08-19 by mightymos