mightymos/RF-Bridge-OB38S003 #6: Communication between OB38S003 and ESP8285

otsoni, 2024-03-22:
Hi,

I have an idea about handling the protocol encoding and decoding on ESP8285 chip and just transferrind the raw data of pulse lengths using uart between those two chips. I'm using esphome and there is already really good libraries for doing the encoding and decoding, but of course it would also need changes to work also with raw data coming from uart. What do you think about that? Is there something that I haven't been thinking and this is just bad idea? I can try to make some kind of pull request if you don't mind, but it may take some time as my C skills aren't very good.

By the way, thanks a lot for your hard work on this!

-Otto

mightymos, 2024-03-23:
Hi Otto,
I have updated the OnbrightFlasher to allow connecting to wifi so that a larger hex file can be uploaded through web browser.
So it should work to flash either the pass through mode or serial mode file in this repository.

I plan to reorganize the source code and learn more about makefiles so that all the firmware modes can be built at the same time.
I have also made progress on porting the original portisch firmware to SDCC compiler:
https://github.com/mightymos/SonOfPortisch

It should hopefully be the next step to allow this to work on the OB38S003.
I hope to have all the options built here so that people can choose.

Have you tried the OnbrightFlasher web upload option recently?
What else do you think?

- Otto

otsoni, 2024-03-23:
I haven't tried the wifi uploader, but I have been using the python script for uploading and it works well for me.

I was thinking to add third mode that is something between pass through and serial modes. So that we don't need to use ESP8285 resources for simple operations like handling the transmission and receiving rf signals. But we could use ESP8285 resources for more demanding operations like encoding and decoding the signals as OB38S003 is somewhat limited for that. And at least for esphome there is already really good encoding and decoding options for rf signals and we could use those and transfer the raw signal data by serial. At least for transmission should work, but I don't know if it would work for receiving in noicy environment.

The idea here is that people could flash the OB38S003 just once and even if they get new devices that need new protocols they don't need to flash the OB38S003 as they are handling the signal encoding and decoding on ESP8285 chip. And I'm not sure if Portisch has support for the devices that I'm using (Nexa), but esphome has. Of course this would need changes to esphome side too.

mightymos, 2024-07-02:
The most straightforward thing is to use the pass through firmware and the remote receiver component in ESPHome:
https://esphome.io/components/remote_receiver.html

The project has an example YAML file for ESPHome now.
Other users here might be able to help to see if your Nexa device is supported:
https://github.com/mightymos/RF-Bridge-OB38S003/issues/8

You can also attempt to decode an unknown protocol using ESPHome dump feature.

closed 2024-07-02 by mightymos