mightymos/RF-Bridge-OB38S003 #9: Tamosta configuration for main_rcswitch_OB38S003.hex

totti001, 2024-07-28:
Hi,

I try to use my Sonof RF bridge V2.2 to clone my RF remote controllers. I flashed the passthrough version, but no success to detect any button press. Now I flashed rcsiwtch version but I could not find which tasmota configuration should be used for that mode. Can you help me? I flashed tasmota_sensors.bin already into the esp8285

Thank you

mightymos, 2024-07-28:
You need to set the module type to Sonoff Bridge (25).

totti001, 2024-07-28:
> You need to set the module type to Sonoff Bridge (25).

And any gpio settings? Ot is there any special setting? like rfraw 177? I tried different commands without any success to detect button press. I have DC306 5 channel remote control.

mightymos, 2024-07-28:
There are no other settings that need to be set if Sonoff Bridge(25) module type is used with rcswitch firmware.
It may just be that the remote uses a different protocol that is unsupported by rcswitch, it is not possible to know without more information.

@the-mentor was helping people to "sniff" their radio protocols to try identifying them. Maybe their Discord server could help.