mightymos/RF-Bridge-OB38S003 #10: RfRaw not working with Tasmota Version 14.2.0

matlab22, 2024-08-25:
I am trying to control my hob (SF01 01319004) fan and LED using the RF bridge. But I can not send raw commands.

```
22:26:25.824 CMD: RfSend 7350492, 24, 1, 10, 238
22:26:25.136 RSL: RESULT = {"RfSend":"Done"}
22:26:31.790 CMD: RfSend 7350492, 24, 1, 10, 238
22:26:31.102 RSL: RESULT = {"RfSend":"Done"}
22:27:17.155 CMD: "AA B1 04 07EB 0157 00FD 3EBC 010101010101101001010101101010100103 55" 20
22:27:47.220 CMD: rfraw 177
22:27:47.226 RSL: RESULT = {"Command":"Unknown"}
22:27:57.594 CMD: RfRaw AAB01C041407EB015700FD3EBC01010101010110100101010110101010010355
22:27:57.601 RSL: RESULT = {"Command":"Unknown"}
22:28:09.902 CMD: RfRaw AAB155
22:28:09.908 RSL: RESULT = {"Command":"Unknown"}
22:28:13.657 CMD: RfSend 7350492, 24, 1, 10, 238
22:28:13.969 RSL: RESULT = {"RfSend":"Done"}
22:28:35.793 CMD: RfSend 1, 1, 1, 10, 999
22:28:35.158 RSL: RESULT = {"RfSend":"Done"}
22:29:12.606 CMD: RfRaw 0xA5/0xA8
22:29:12.612 RSL: RESULT = {"Command":"Unknown"}
22:29:16.708 CMD: RfRaw 0xA5
22:29:16.714 RSL: RESULT = {"Command":"Unknown"}
22:29:22.788 CMD: RfRaw 0xA5
22:29:22.794 RSL: RESULT = {"Command":"Unknown"}
22:29:28.036 CMD: RfRaw 0xA5
22:29:28.043 RSL: RESULT = {"Command":"Unknown"}
22:29:29.616 CMD: RfRaw 0xA5
22:29:29.622 RSL: RESULT = {"Command":"Unknown"}
22:29:30.892 CMD: RfRaw 0xA5
22:29:30.899 RSL: RESULT = {"Command":"Unknown"}
22:29:32.991 CMD: RfRaw 0xA5
22:29:32.998 RSL: RESULT = {"Command":"Unknown"}
22:29:44.665 CMD: RfSend 1, 1, 1, 10, 999
22:29:44.031 RSL: RESULT = {"RfSend":"Done"}
22:30:44.454 CMD: RfSend 1, 1, 8, 10, 999
22:30:44.860 RSL: RESULT = {"RfSend":"Done"}
22:30:52.873 CMD: RfSend 1, 1, 8, 10, 999
22:30:52.279 RSL: RESULT = {"RfSend":"Done"}
```

I would like to send those commands:
PLUS
1011011001001001001011011001001001001001011001001011011
b6492d924964b6

LIGHT
1011011001001001001011011001001001001001011001001001011
b6492d92496496

TIMER
1011011001001001001011011001001001001001011001011011001
b6492d924965b2

MINUS
1011011001001001001011011001001001001001011001011001011
b6492d92496596

FAN
1011011001001001001011011001001001001001011001001011001
b6492d924964b2

ASK
55bit
Duration of 1 bit (1): 343 µs
Duration of 0 bit (0): 391 µs
Total sequence time: ~19.35 ms
Repetition: 10 times, with a pause between sequences.

how can I achieve this? It looks like raw commands are not working

mightymos, 2024-08-26:
I only got the 0xA5 transmit command working which expects 24 bits of data:
https://github.com/Portisch/RF-Bridge-EFM8BB1/wiki/0xA5

(I also added 0xA8 but I do not have a way to test it because none of my devices use it).

It looks like you are needing to send 55 bits (56 bits or 7 bytes?) of data.
It might just be best to flash the passthrough firmware and then work with ESPHome:
https://esphome.io/components/remote_transmitter.html
remote_transmitter.transmit_raw Action

mightymos, 2024-08-28:
I double checked and Tasmota is using the rcswitch library itself and does not appear to allow sending raw:
https://tasmota.github.io/docs/RF-Protocol/

If possible in the future I may try to include these Portisch features in this firmware.
However, it may never happen as most people seem to be inclined to just flash the passthrough and use Tasmota if their devices are controllable by it or ESPHome otherwise.

closed 2024-08-28 by mightymos

matlab22, 2024-08-30:
I switched to a ESP32C3 and CC1101 with ESPhome:
```
button:
  - platform: template
    name: Light
    on_press:
      - lambda: get_cc1101(transciver).beginTransmission();
      - remote_transmitter.transmit_raw:
          # prettier-ignore
          code: [300, -300, 600, -300, 600, -600, 300, -600, 300, -600, 300, -600, 300, -300, 600, -300, 600, -600, 300, -600, 300, -600, 300, -600, 300, -600, 300, -300, 600, -600, 300, -600, 300, -600, 300, -300, 600]
          repeat:
            times: 10
            wait_time: 15.15ms
      - lambda: get_cc1101(transciver).endTransmission();
```

you can find my progress here:
https://github.com/matlab22/RF_Hob_Control_SF0101319004

to get the raw signals for esp home you can use the python script:
https://github.com/matlab22/RF_Hob_Control_SF0101319004/blob/main/Convert-HexToESPHomeRaw.py

mightymos, 2024-08-30:
That's really cool and probably equivalent effort or even easier than hacking on the Sonoff.
I'm glad ultimately the fan is able to be controlled.

I'll try to remember to add a link for related projects in my readme so people know they have options.
If I forget, feel free to remind me in case they don't see the links here.

matlab22, 2024-08-31:
The sad thing I had the CC1101 before I bought the sonoff. I did not find an easy way to use the CC1101 with a raspi so decided to by the sonoff. then they made I new version and I invested time in hacking this, to find out my fan uses a special protocol. So I bought the RTL-SDR to sniff it. Then I found out, the output of the sonoff is crap and switched to the CC1101 and had it running under 2h.
Just a side question, before I trough the sonoff away. Is it possible to change the carrier frequency on the sonoff?

mightymos, 2024-08-31:
I'm not aware of a way to control the carrier frequency with software.
In hardware, as I vaguely recall from the datasheets to change frequency on these you have to use a different part number.

Possibly you could find someone locally to you that could use the Sonoff.