mightymos/RF-Bridge-OB38S003 #12: Issues with ESP after flashing the firmware to a 433Bridge R2 V2.2

leecher1337, 2024-10-03:
I don't know what I did wrong, I followed the tutorial and flashed the Passthrough-firmware to the Sonoff 433Bridge R2 V2.2 (2021.01.19):

```
>python flashScript.py
Available COM ports:
1. COM1 - Kommunikationsanschluss (COM1)
2. COM6 - USB-SERIAL CH340 (COM6)
Select a COM port (enter the number): 2
2024-10-03 18:55:27,168 - INFO - Serial connection opened on COM6
Select a file by number:
1. blink.ihx
2. main_passthrough_OB38S003.hex
Enter the number of the file you want to select: 2
2024-10-03 18:55:29,211 - INFO - You selected: main_passthrough_OB38S003.hex
2024-10-03 18:55:30,315 - INFO - handshake
2024-10-03 18:55:30,316 - INFO - State changing to handshake
2024-10-03 18:55:30,318 - INFO - cycle power to target (start with power off and then turn on)
2024-10-03 18:55:33,812 - INFO - Handshake succeeded
2024-10-03 18:55:33,930 - INFO - Status: 0
2024-10-03 18:55:33,931 - INFO - Success
2024-10-03 18:55:33,933 - INFO - Chip read: 0xA
2024-10-03 18:55:33,934 - INFO - Connected...
2024-10-03 18:55:33,937 - INFO - Returning to idle state...
2024-10-03 18:55:34,038 - INFO - erase
2024-10-03 18:55:34,539 - INFO - Erasing chip...
2024-10-03 18:55:34,540 - INFO - Status: 0
2024-10-03 18:55:34,541 - INFO - Success
2024-10-03 18:55:34,542 - INFO - Chip erase successful
2024-10-03 18:55:34,643 - INFO -
2024-10-03 18:55:35,143 - INFO - setfuse 18 249
2024-10-03 18:55:35,644 - INFO - Set configuration byte...
2024-10-03 18:55:35,645 - INFO - Status: 0
2024-10-03 18:55:35,646 - INFO - Success
2024-10-03 18:55:35,647 - INFO - Wrote configuration byte
2024-10-03 18:55:35,653 - INFO - :03000000020006F5
2024-10-03 18:55:35,656 - INFO - Write successful
2024-10-03 18:55:35,660 - INFO - Wrote 3 bytes
2024-10-03 18:55:35,673 - INFO - :09005F0075080075090002000398
2024-10-03 18:55:35,681 - INFO - Write successful
2024-10-03 18:55:35,684 - INFO - Wrote 9 bytes
2024-10-03 18:55:35,689 - INFO - :0300030002008870
2024-10-03 18:55:35,692 - INFO - Write successful
2024-10-03 18:55:35,694 - INFO - Wrote 3 bytes
2024-10-03 18:55:35,704 - INFO - :10006800D2809000141200E4C28022D2B09003E83B
2024-10-03 18:55:35,720 - INFO - Write successful
2024-10-03 18:55:35,722 - INFO - Wrote 16 bytes
2024-10-03 18:55:35,733 - INFO - :100078001200E4C2B0D2B09003E81200E4C2B02289
2024-10-03 18:55:35,748 - INFO - Write successful
2024-10-03 18:55:35,751 - INFO - Wrote 16 bytes
2024-10-03 18:55:35,759 - INFO - :1000880012010812012AC2B0C280C287C290C29768
2024-10-03 18:55:35,774 - INFO - Write successful
2024-10-03 18:55:35,777 - INFO - Wrote 16 bytes
2024-10-03 18:55:35,784 - INFO - :100098001200739001F41200E4D2B012010C1201A4
2024-10-03 18:55:35,800 - INFO - Write successful
2024-10-03 18:55:35,803 - INFO - Wrote 16 bytes
2024-10-03 18:55:35,810 - INFO - :1000A80026AF08A296E433F508EFB50802800AE502
2024-10-03 18:55:35,826 - INFO - Write successful
2024-10-03 18:55:35,828 - INFO - Wrote 16 bytes
2024-10-03 18:55:35,836 - INFO - :1000B800086004D2908002C290AF09A291E433F59F
2024-10-03 18:55:35,852 - INFO - Write successful
2024-10-03 18:55:35,854 - INFO - Wrote 16 bytes
2024-10-03 18:55:35,862 - INFO - :1000C80009EFB5090280D7E5096004D28780CFC25D
2024-10-03 18:55:35,877 - INFO - Write successful
2024-10-03 18:55:35,880 - INFO - Wrote 16 bytes
2024-10-03 18:55:35,888 - INFO - :1000D8008780CBAF828F061FEE70FA22AE82AF8385
2024-10-03 18:55:35,903 - INFO - Write successful
2024-10-03 18:55:35,906 - INFO - Wrote 16 bytes
2024-10-03 18:55:35,914 - INFO - :1000E8007C007D00C3EC9EED9F50147A957B051A29
2024-10-03 18:55:35,929 - INFO - Write successful
2024-10-03 18:55:35,932 - INFO - Wrote 16 bytes
[... 78 lines cut ...]
2024-10-03 18:55:36,587 - INFO - :10003B007800E84400600A790175A000E4F309D860
2024-10-03 18:55:36,602 - INFO - Write successful
2024-10-03 18:55:36,604 - INFO - Wrote 16 bytes
2024-10-03 18:55:36,612 - INFO - :10004B00FC7800E84400600C7900900001E4F0A318
2024-10-03 18:55:36,627 - INFO - Write successful
2024-10-03 18:55:36,630 - INFO - Wrote 16 bytes
2024-10-03 18:55:36,636 - INFO - :04005B00D8FCD9FAFA
2024-10-03 18:55:36,640 - INFO - Write successful
2024-10-03 18:55:36,642 - INFO - Wrote 4 bytes
2024-10-03 18:55:36,655 - INFO - :0D00060075810D12024BE5826003020003BC
2024-10-03 18:55:36,668 - INFO - Write successful
2024-10-03 18:55:36,670 - INFO - Wrote 13 bytes
2024-10-03 18:55:36,676 - INFO - :04024B007582002296
2024-10-03 18:55:36,680 - INFO - Write successful
2024-10-03 18:55:36,682 - INFO - Wrote 4 bytes
2024-10-03 18:55:36,687 - INFO - :00000001FF
2024-10-03 18:55:36,787 - INFO - mcureset
2024-10-03 18:55:37,288 - INFO - MCU reset...
Serial connection closed.
```

After flashing completed, the RED led turned on.
Since then, I'm unable to flash the ESP8266 on the board, as soon as I apply power to the Sonoff, even though pressing and holdinig the button (GPIO0), LED immediately turns on and stays on, so that I'm unable to flash any firmware to the device.

Do you have a clue what went wrong here?

mightymos, 2024-10-03:
How are you powering the Sonoff?
From what I recall it has to be 3.3V powered (5V USB power would not work), see instructions here:
https://tasmota.github.io/docs/Getting-Started/#hardware-preparation

And what are you using on PC side to flash ESP8285?
What error message does it give?

leecher1337, 2024-10-03:
I'm powering it via 3.3V from the Serial adapter. Before flashing the RF-Bridge-OB38S003 firmware, I already flashed Tasmota to it successfully and also did the Erase.
I tried with Tasmotizer and with Tasmota Web install, all say that they cannot communicate with the Chip.
Normally, when pulling GPIO 0 to GND on startup, the chip should remain silent and wait for flash commands, but after flashing this firmware, suddenly the red LED turned on and now it stays on as soon as I apply power to the board. 
Maybe the firmware is spamming the bus as soon as it gets loaded or something like that?
Did I select the wrong firmware?

mightymos, 2024-10-03:
If you already flashed Tasmota then I do not understand why you are trying again.
Also I do not know what firmware you flashed to the microcontroller.

I believe there is a series resistor on the transmit and receive traces connecting the ESP8285 and microcontroller.
So driving them on the ESP side connector should still work regardless (J2 connector).
However, it's at least possible that serial lines could interfere.

Make sure your adapter is connected to J2 connector.

Alternatively, you can erase flash on microcontroller so you know no code is running.
Then flash ESP with Tasmota.
Then reflash microcontroller with HEX.

leecher1337, 2024-10-03:
> If you already flashed Tasmota then I do not understand why you are trying again. Also I do not know what firmware you flashed to the microcontroller.

I had to erase Tasmota from the board again, as the instructions said so (in order to not interfere with the Microcontroller, it was recommended to erase any firmware which I did, so I removed Tasmota again before doing the procedure).

Of course, I used J2 connector for flashing Tasmota. But the hint that the Red led is from RF (as already written on the board, I should have noticed that before) at least tells me that I shouldn't pay attention to it regarding bootup status.

Is it normal that RED light fires up immedaitely after boot using your firmware? To be more precise, it turns red immediately, then it turns off for for approx 0.2 sec  and then starts again and from this point on it is permanently on.

I can try what you suggested to remove firmware from RF controller again and see if I can flash Tasmota then, still the question arises: As it is recommended to do the flash only after erasing main firmware from the ESP, would flashing still work with Tasmota firmware on the ESP without interference?

mightymos, 2024-10-03:
I looked again at the passthrough source code and it shows two short LED blinks, followed by a delay, and then the LED is set to on, so yes that sounds normal.

I guess you should have left Tasmota on the ESP8285, it cannot really interfere with flashing microcontroller since that flash happens over normally unconnected I2C pins. However, it is possible microcontroller can interfere with ESP flashing  because of sharing serial pins.

So the order should be erase ESP, flash Tasmota (or similar), then erase microcontroller, then flash microcontroller.

It would be nice if the hardware was arranged more like a development board with reset pins accessible/controllable etc but as is it is cumbersome.

If you can point me to where is the suggestion to erase Tasmota, I will look at it, it seems like it is bad advice.

leecher1337, 2024-10-03:
Thanks for your fast reply, I'll try it out.

Advice is from: https://github.com/mightymos/OnbrightFlasher/blob/main/flashing-guide-by-example.md#erasing-the-esp-on-the-pcb

leecher1337, 2024-10-04:
So from what I tested, I can say that the OB38S003 firmware really seems to have blocked ESP flshing.
As soon as I deleted the RF-Bridge-OB38S003 firmware with `erase` command, I was able to flash Tasmota again.
So I did as advised and flashed the passthrough mode firmware afterwards. Now I have Tasmota running and applied template

`{"NAME":"RFB v2.2","GPIO":[1,1,1,1,1152,1120,1,1,1,288,1,1,1,1],"FLAG":0,"BASE":18}`

(GPIO4 RfREcv, GPIO5 RfSend)
Tried to receive RF signal from a remote but nothing can be seen in the logs:

```
00:00:22.383 CMD: SerialLog 0
00:00:22.386 RSL: RESULT = {"SerialLog":{"0":{"Active":"0"}}}
00:00:29.655 CMD: RfProtocol a
00:00:29.663 RSL: RESULT = {"Protocol":"1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37"}
```

Should I try the other firmware, which directly decodes on the µC, instead or did I miss something?

leecher1337, 2024-10-04:
Sorry for the noise, I used another RF remote now and used GPIO1 and GPIO3 (For the record, template is: `{"NAME":"RFB v2.2","GPIO":[1,1120,1,1152,1,1,1,1,1,288,1,1,1,1],"FLAG":0,"BASE":18}` and it received signals sometimes (only about 70% of all the time compared with other receivers that I tested). I guess it's as good as it can get.

mightymos, 2024-10-04:
> Sorry for the noise, I used another RF remote now and used GPIO1 and GPIO3 (For the record, template is: `{"NAME":"RFB v2.2","GPIO":[1,1120,1,1152,1,1,1,1,1,288,1,1,1,1],"FLAG":0,"BASE":18}` and it [...]

No problem, yes those are the correct pins.
I don't know anything about the templates but you might consider using the one provided here (or submitting your template to that website since it applies to R2.2 hardware version):
https://templates.blakadder.com/sonoff_RF_bridge.html

It is unlikely my firmware rcswitch would be better than tasmota rcswitch because they are both based on the same decoding principle (i.e. comparing measured timings to known protocols). Passthrough seems to be preferred by most people because many more protocols are supported by tasmota due to more memory availability along with more features.

Usually the problem people have is that they do not know the protocol their fan, blinds, device, etc. is using and therefore do not know if they are receiving a valid code or not.

But in your case it's too bad it seems to be working but unreliable.

Can you post your good codes or just everything shown in your tasmota console and tell me which codes are good so I can possibly understand further?

zd3sf, 2024-11-06:
I only had luck flashing the ESP8266 when I power it using the USB microB.

mightymos, 2024-11-23:
@leecher1337 

Another user has had good success with sniffing remote codes using the Portisch firmware in the new release:
https://github.com/mightymos/RF-Bridge-OB38S003/releases/tag/v0.4.4

I know it is cumbersome to flash new versions, but if you flash Portisch, issue "rfraw AA B1 55" in Tasmota console interface to enter sniffing mode, then press buttons on your remote and post the output from Tasmota here we can hopefully help you to replay them so as to behave like your original remote.

leecher1337, 2024-11-23:
Thank you very much for you kind offer for help! I did the flash for a friend of mine who owns the device and I need to get hold of it again first, that's why I didn't respond yet. Having portisch supported now in your new release looks very interesting, as that's the firmware I basically wanted to substitute for the new RF chip on this hardware.
I'll try as soon as I have the device again.

mightymos, 2024-11-28:
Closing for now since readme instructions for flashing have been clarified.
Please open issues as needed if testing reveals problems.

closed 2024-11-28 by mightymos

leecher1337, 2024-12-20:
Hi,
Just wanted to report back. Portisch Firmware looks good, used it successfully on the target device now. 
Thank you!

mightymos, 2024-12-20:
> Hi, Just wanted to report back. Portisch Firmware looks good, used it successfully on the target device now. Thank you!

Great, that's good to note.