mightymos/RF-Bridge-OB38S003 #13: A few questions

felipecrs, 2024-11-06:
Hi there,

I was looking for a Portisch firmware replacement (which is no longer being developed) for my SONOFF RF Bridge R2 V1.0 and found yours.

If you don't mind, I have a few questions.

1. What is the relationship of this with the Portisch firmware? Is this a "fork"/"continuation"?
3. What does this "Portisch: No" in the README table means?
2. Status says: "should not be used by typical users", but is this also valid for EFM8BB1 or only for OB38S00?

Thanks a ton.

mightymos, 2024-11-06:
Hi, I recently modified Portisch source to compile with SDCC compiler and to use hardware abstraction.
Unfortunately, Portisch does not yet work on OB38S003 for unknown reasons despite being able to compile.
I hope to figure out why sooner or later.

However, since you have the Sonoff R2 V1.0 (black box correct?), you can test this Portisch port here:
https://github.com/mightymos/RF-Bridge-OB38S003/releases/tag/v0.4.0

The rcswitch version is basically just a smaller compiled size so more protocols can be supported at one time.
Portisch takes up more code space and therefore cannot fit many protocols.
Howver Portisch has the sniffing feature - I can produce output from my radio door sensor durign sniffing but I do not know if the output is actually correct or ultimately helpful to decode unknown protocols.
The passthrough version basically just allows decoding on the ESP8285 itself without requiring hardware modification.

I have only tested these firmware primarily on my development board, so if you try them out on black box Sonoff it would be helpful to hear if you are successful in decoding or transmitting or not.

felipecrs, 2024-11-06:
That sounds very good. I don't think I will be able to test it anytime soon, but if I do, I'll surely report back.

Thank you very much for the response, and good luck with the project.

closed 2024-11-06 by felipecrs

mightymos, 2024-11-23:
This single but thorough user has managed to test the Portish port with a good deal of success:
https://github.com/mightymos/RF-Bridge-OB38S003/issues/11

However, I really can't say there is a particular reason to use this version for v1.0 hardware users unless some type of bug(s) are found which was not previously addressed in original Portisch...
https://github.com/Portisch/RF-Bridge-EFM8BB1/issues/232

And even then I am not sure if any bug is visible to user operation or not.
Anyway, cheers mate.