mightymos/RF-Bridge-OB38S003 #19: Sonoff RF bridge OB38S003 Portisch FW freezes after 24-48 hours

zd3sf, 2024-11-30:
For receive only. Transmit keeps working fine!

Sonoff RF bridge V2.2 with [v0.4.5] portisch firmware and ESPHome 2024.11.2

After 24-48 hours of continuous operation, the MCU stops receiving 0xA4 codes. Restarting the ESP doesn't resolve the problem. 

Workaround: reset MCU using rfraw AA FE 55, start bucket sniffing then stopping, or power unplug. MCU can be reset on a schedule using home assistant or ESPHome.

Logs: none yet, just changed ESPHome log level to Debug and will record results in an update.

otsoni, 2024-12-03:
I got some problems just after receiving one long code (66 bytes, I think). It seems to keep receiving, but beeping causes issues. This might be something else than your problem but I will dig into my problem more and hopefully we can find resolution to both problems. I already added stack trap for the last two bytes of internal RAM and it seems like it is not stack overflow, but not really sure about that. I have plans to add traps after arrays in external RAM to see if those are overflowing

otsoni, 2024-12-03:
I think that my problem with beeping after receiving large bucket is some other bug as it seems that nothing is overflowing. I need to add more traps and keep board running until it stops receiving buckets and then check the traps. It will take some time

otsoni, 2024-12-04:
I found the beep bug. https://github.com/mightymos/RF-Bridge-OB38S003/blob/main/src/main_portisch.c#L880 should be `(RF_DATA[1] << 8) | RF_DATA[0]` instead of `(RF_DATA[2] << 8) | RF_DATA[1]`

I think that I found the B1 overflow too, but not sure about that as it is working with the memory traps, but memory trap after buckets (https://github.com/mightymos/RF-Bridge-OB38S003/blob/main/src/portisch.c#L53) is changed. I think that the check here https://github.com/mightymos/RF-Bridge-OB38S003/blob/main/src/portisch.c#L729-L733 should be done before writing the duration on row 724. For my understading it is now checking the overflow condition after the overflow has already happened, but correct me if I have been thinking something silly.

It seems that there is same kind of bug with RF_DATA (https://github.com/mightymos/RF-Bridge-OB38S003/blob/main/src/portisch.c#L737-L755) if I understand the logic correctly. I haven't tried those changes yet.

I tried to understand from the memory map what happens with the overflow. It seems to overwrite bucket_sync variable and I just don't understand why that disables the B1 listening. Maybe I don't need to understand everything. Or maybe my build wiht the memory traps is mapping the memory in a different way than the original.

zd3sf, 2024-12-05:
I'm sorry, this goes beyond what I know about these microcontrollers. I should also report that I have strange thing happen where my window blind (B0 transmit) goes up on its own without anyone's input. I check ESPHome logs and I find nothing.

mightymos, 2024-12-05:
> I'm sorry, this goes beyond what I know about these microcontrollers. I should also report that I have strange thing happen where my window blind (B0 transmit) goes up on its own without anyone's input. I check ESPHome logs and I find nothing.

@otsoni thanks for the catch on the beep, I think you are right and I think I fixed it. I do not have the buzzer installed on my board so I do not actually hear any beep.

I think I made a mistake previously with setting state for the state machines in the main file instead of between two files. I think it is back to the original portisch steps, but I just did a quick check that hardware could send, receive, etc, no long time period checking.

I will eventually need to set the state in one file because it is too confusing to follow the logic when it is set between two files. But I am hoping by making it like original portisch for now testing can be more stable. Sorry for all the testing need. I will make the code more readable later, new release is published.

zd3sf, 2024-12-05:
It was not in sniffing mode, just in standard decoding. I am not sure why the bridge is sending commands on its own, no one is touching the physical remote.

mightymos, 2024-12-06:
> Thanks guys for looking into it. The bridge was not in sniffing mode, just in standard decoding. I am not sure why the bridge is sending commands on its own, no one is touching the physical remote either.

I just observed I receive 00C303 codes on my original portisch black box after flashing a new yaml to the white sonoff.
When I flash the yaml over wifi it takes awhile for the logging to connect and so I do not know if I am missing messages or not.

My guess is that ESPHome is executing a default action on startup so that the visual toggle in HA matches the last executed action.
Possibly something similar is happening with your blinds.

However, I am not yet ready to blame ESPHome until this behavior can be confirmed.

otsoni, 2024-12-07:
> I think that I found the B1 overflow too, but not sure about that as it is working with the memory traps, but memory trap after buckets [...]

I still think that this should fix the code receiving problem. I can't make a build for you with those fixed as for some reason I get `Insufficient ROM/EPROM/FLASH memory` error when building the main branch. @mightymos have you done something special to get the builds or is there something wrong in my build environment?

I think that I could try to use other rf-bridge in passthrough mode to send code with too many buckets and test if I can get other rf-bridge to freeze straight away.

zd3sf, 2024-12-07:
1) I figured out the blinds moving on their own. I have an automation that sets the blind position at 40%, if it gets triggered and the blinds are already at 40%, there's a bug where ESPHome tries to move the blinds for a millisecond then stops. But the problem is that with RF, the  UP/Down commands take some time, that the stop command is issued while the bridge is busy, so it never sends. See ESPHome bug [here](https://github.com/esphome/issues/issues/1061). I'll set a condition where the automation doesn't get executed if set position = current position. 

2) Bridge still freezes. I have a reset command automation every hour, but every now and then, I see a freezing condition. No big deal but just reporting  it.

otsoni, 2024-12-07:
Nice that you foud out the bug. Next I will try to make some kind of radio signal that causes the bridge to freeze. I think it should have at least 9 different  bucket lengths. I hope that I can create it with passthrough firmware and esphome. I let you know when I have some more information about that.

mightymos, 2024-12-07:
Support for learning mode has been added. Portisch port should be feature complete with original now.
To support learning mode I had to fix the timer to work with both microsecond and millisecond delays.

I do not fully understand it, but Portisch computes a checksum along with an 800 millisecond delay with repeat codes.
So as a result of the timer fix, you will see only one code even when repeats are sent and this is how the original behaved.
rcswitch just decodes as fast as it can and so repeats are still output.

I also made some organizational changes to make state machines more readable for me.
I plan to stop making further changes because it is probably providing marginal improvement and the code needs to finally "freeze" to exhaustively investigate freezing or other bugs.

@otsoni If you are making changes to the code you need to monitor the .MEM file to make sure you are not going over microcontroller memory limits (ram, xram, flash). Thanks for trying out your testing strategy, it is difficult for me to test on my own.

@zd3sf Thanks for the additional work with the blinds. I know the example yaml you have is probably diverging from my own. If you want to somehow consolidate changes later or post your own custom yaml file I can include it in the example folder and not edit it once it seems complete.

zd3sf, 2024-12-07:
Thanks @mightymos, we can merge the YAMLs at some point. most of the changes are situation-specific and not related to the general operation of the bridge. I added one line `restore_from_flash: true`. This allows the state of the blinds to persist between reboots of the ESP. 

```
esp8266:
  board: esp01_1m
  restore_from_flash: true
```

I tried the latest firmware, standard and bucket recieve work as intended. But, I couldn't get transmit to work at all standard or Bucket. I tried re-sniffing the codes and transmit them, but didnt work.

otsoni, 2024-12-08:
> @otsoni If you are making changes to the code you need to monitor the .MEM file to make sure you are not going over microcontroller memory limits (ram, xram, flash). Thanks for trying out your testing strategy, it is difficult for me to test on my own.

I have been monitoring the .MEM file but it gives me the error also when building without any changes to the main branch. I will create separate issue about that.

otsoni, 2024-12-08:
> I have been monitoring the .MEM file but it gives me the error also when building without any changes to the main branch. I will create separate issue about that.

My bad. It seems that my Ubuntu WSL had some old version of sdcc (I think it was 4.0.0) in it's repositories and updating to 4.4.0 seems to fix build problems. I will continue to debug the freezing problem

otsoni, 2024-12-08:
@zd3sf I just realized that you have problems with 0xA4 codes freezing and I have been using just 0xB1 sniffing. It seems that the problems with overflowing buffers are not related to 0xA4, just 0xB1 sniffing. So there might be two separate freezing problems. (Or one that is the cause for both and I have been tracing just wrong things)

mightymos, 2024-12-09:
> Thanks @mightymos, we can merge the YAMLs at some point. most of the changes are situation-specific and not related to the general operation of the bridge. I added one line `restore_from_flash: [...]

I made a new release to declare feature complete with learning mode.

During development I used one of the hardware timers for software uart to output debug information.
But this is usually unnecessary now since I could use the busybee kit instead.
So ironically I was basically undoing work I had done previously.

Anyway, using two timers for delays was the correct architecture choice.
I appear to be able to decode, sniff, transmit using esphome to original sonoff receiver.

If it is still not working with your devices I will need to hookup oscilloscope to check signal timings.
But I wanted to post a fixed version.

If you could once again be patient and try it that would help, I'll hope for good news.

mightymos, 2024-12-09:
> My bad. It seems that my Ubuntu WSL had some old version of sdcc (I think it was 4.0.0) in it's repositories and updating to 4.4.0 seems to fix build problems. I will continue to debug the freezing [...]

Ah okay, good catch.
Yes I use version 4.4.0 on a Windows machine natively.

zd3sf, 2024-12-10:
Okay testing 0.4.8 standard transmit/receive working fine. For door sensor, I revive 10 signals each trigger (like rcswitch) , while the broken 0.4.7 would receive only one due to the checksum timer.

B0/B1 working. I can control the blinds. Like previous versions, Bucket sniffing sniffs one code then stops. You'd have to reset the mcu and start sniffing again. Doesn't bother me because I only need the codes for the blinds.

I disabled the mcu reset automation from home assistant to see if the mcu will freeze.

zd3sf, 2024-12-10:
Freezing happened overnight. Standard and B0 transmit remained working, but standard receive stopped working until an MCU reset.

mightymos, 2024-12-16:
> Freezing happened overnight. Standard and B0 transmit remained working, but standard receive stopped working until an MCU reset.

After all that I had actually misread portisch source and implemented the delays incorrectly.
The hardware timers on the older silicon labs microcontroller are more advanced than the new onbright microcontroller, so it is not completely my fault.
It should finally be fixed now in v0.4.9.

The only noticeable difference is that checksum timer should ignore outputting repeat codes.
I think everything is still working.

I plan to finally pursue freezing in earnest now.
I'll leave my hardware running continuously as well and see what I can observe.

zd3sf, 2024-12-17:
Alright, just flashed 0.4.9. I like the repeats because the feedback is instantaeous (ex: open the backdoor and the lights come on). But I would like to see if maybe the freezing is due to the MCU having to output all the repeats. Testing now without an auto reset script. 

The timer delay doesnt feel too different from before testing it on the backdoor lights.

mightymos, 2024-12-20:
> Alright, just flashed 0.4.9. I like the repeats because the feedback is instantaneous (ex: open the backdoor and the lights come on). But I would like to see if maybe the freezing is due to the MCU having to output all the repeats. Testing now without an auto reset script.
> 
> The timer delay doesn't feel too different from before testing it on the backdoor lights.
> 
> Update: It froze after 15 hours on V0.4.9

Ok, my unit keeps working for days with me occasionally clicking radio buttons etc.
The problem must arise because you are more actively using the bridge.
I am able to reconnect to logging over wifi this way apparently:
```
esphome logs
```

Do you have logs with uart bytes that we can use to narrow down when the problems begin?

mightymos, 2024-12-21:
Okay, I had a "freeze" with red LED solid on that could only be remedied by cycling power from msm9096 flasher.
Unfortunately logging in Powershell was not running.

I am trying to run the bridge again with logging on continuously.

mightymos, 2024-12-27:
> Were you logging using the Onbright flasher? Is there a way to log through ESPHome?

For ESPHome, you need the Logger component like this:
```
# Enable logging
logger:
  baud_rate: 0
```

The baud_rate zero actually DISABLES logging output from UART as I recall but logging still works over wifi.

So then you can do:
```
esphome run example.yaml
```

or to reconnect another time:
```
esphome logs example.yaml
```

You also need this to see raw uart bytes:
```
uart:
  tx_pin: 1
  rx_pin: 3
  baud_rate: 19200

  # this shows raw bytes in log (can be viewed in terminal over wifi)
  debug:
```

I use Powershell on Windows.
I use Insomnia to keep the computer from sleeping:
https://dlaa.me/Insomnia/

zd3sf, 2024-12-28:
I have both, but I see nothing about the freezing in ESPHome logs (using home assistant Add-on )

```
# Enable logging
logger:
  baud_rate: 0
  level: DEBUG

 
uart:
  tx_pin: 1
  rx_pin: 3
  baud_rate: 19200
```

mightymos, 2024-12-28:
> I have both, but I see nothing about the freezing in ESPHome logs (using home assistant Add-on )
> 
> ```
> # Enable logging
> logger:
>   baud_rate: 0
>   level: DEBUG
> 
>  
> uart:
>   tx_pin: 1
>   rx_pin: 3
>   baud_rate: 19200
> ```

Well there is not a particular message, but hopefully we could catch bytes being sent and received with valid formats until it is not working and that would be some kind of clue as to what is ultimately triggering a problem.

otsoni, 2025-01-14:
Is the freezing still happening to you guys? I realized that in my system esphome dashboard goes offline every night as my computer goes to sleep. In esphome api component there is default 15 minute timeout for missing API connection and after the timeout it reboots the esp-chip and it seems to reset also the mcu. For me that was the reason for mcu stopping advanced sniffing. After adding `reboot_timeout: 0s` under api in esphome yaml my rf-bridge has been receiving advanced codes for five days at least.

mightymos, 2025-01-15:
> For me that was the reason for mcu stopping advanced sniffing. After adding `reboot_timeout: 0s` under api in esphome yaml my rf-bridge has been receiving advanced codes for five days at least.

Despite intending to I have not looked into the freezing much.
That is an interesting finding regarding reboot_timeout, I will have to check it out once there's time.

zd3sf, 2025-01-16:
responding late because I was doing testing. Killed the automation that resets the OB38S003, and it froze in 19 hours. No signs on the logs.

WilldabeastHA, 2025-01-19:
I have similar issue. Portisch FW v2.2 bridge with esphome. The MUC locks up after a couple days and doesn’t revive rf codes. The restart radio button does make it operational again. 

Any suggestions or possible to load a different FW?

mightymos, 2025-01-19:
> I have similar issue. Portisch FW v2.2 bridge with esphome. The MUC locks up after a couple days and doesn’t revive rf codes. The restart radio button does make it operational again.
> 
> Any suggestions or possible to load a different FW?

Another user suggested this for esphome yaml to fix the issue (under api:) =>
```
reboot_timeout: 0s
```

I am trying it out but I do not know for sure if it will work.
You could also setup an automation to periodically trigger a reset of the MCU (e.g., every 8 hrs).

That would use this portion of the example yaml:
```
  - platform: template
    name: "restart radio"
    id: mcu_reset
    on_press:
      then:
      - rf_bridge.send_raw: 
          raw: 'AAFE55'
```

zd3sf, 2025-01-19:
I am testing this solution now and will report back.

WilldabeastHA, 2025-01-24:
> I am testing this solution now and will report back.
> 
> Edit: Bridge froze in 12 hours even when 'auto_reboot= 0'. Will try the new merge to fix [v0.4.11](https://github.com/mightymos/RF-Bridge-OB38S003/tree/v0.4.11) and report back
> 
> Edit2: 0.4.11 froze too. Trying 0.4.12

Any luck with the updated firmware? 

I haven't tried the automation to reset the MCU every 8 hours yet. I’d like to avoid this if possible tho I think it could work.

zd3sf, 2025-01-25:
0.4.12 crashed as well. Automation to reset works fairly well. But the reset takes 1-2 seconds in which my alarm system will be down,

WilldabeastHA, 2025-01-25:
Oddly enough I was gone this week, left Monday came back Saturday. Expected it to not work and it’s working and picking up rf codes. Not sure why that is or for how long it stalls out for and if it comes back. Will have to test and see how long it will black out for and if it comes back on its own.

mightymos, 2025-01-29:
Advanced decoding (0xA6) appears to be working on both boards.
However, in summary it is temperamental with one of my devices while working with others.

I produce a firmware version now that includes sniffing mode, and another that omits sniffing to allow inclusion of multiple protocols.
So hopefully at least a few other people can confirm some devices work.

I have a new BB52 development kit. It has significantly more code and RAM space.
I may port to this board and use the second UART to attempt to log buffer overflows.

zd3sf, 2025-02-02:
I'm still on 0.4.12 but I haven't seen any crashes in the last 4 days. I changed 2 things 1) Debug level to very verbose 2) making the bridge beep when a code is received. Not sure which did it, but I wanted the beeping since the bridge is part of an alarm system now.

mightymos, 2025-02-05:
> I'm still on 0.4.12 but I haven't seen any crashes in the last 4 days. I changed 2 things 1) Debug level to very verbose 2) making the bridge beep when a code is received. Not sure which did it, but I wanted the beeping since the bridge is part of an alarm system now.

Could you try to issue a reset during on_boot (I guess alternatively you could issue a beep or both reset and beep):
https://esphome.io/components/esphome.html

And does issuing a beep periodically seem to solve the issue as well or no (e.g., every 4 hours)?

If I was designing this hardware from scratch I would allow the ESPx to reset the microcontroller by pin.

WilldabeastHA, 2025-02-05:
Mine eventually crashed and doesn’t come back, needed to reset mcu to get it receiving codes again. Will upgrade to 0.14.2 - Will not have it beep on code receive but will see how stable it is and report back. I want to avoid creating an automation to reset the mcu after some period of time because it could be out for an extended period of time.

mightymos, 2025-02-16:
@zd3sf 

Using the BB52 development kit with two UARTs I identified an overflow in the UART state machine.
If for example a too long 0xB0 command is sent, afterward decoding stops blinking the LED or providing decoded output.

Issuing a reset (0xFE) will fix the issue.

I need to think about how to fix the issue in software.

Is this an actual 0xB0 command you use or did I copy and past something incorrectly?
https://github.com/mightymos/RF-Bridge-OB38S003/blob/a481005bbb72c84fe38d43c1c3e87d989aea1c88/example_esphome_yaml/sonoff_rcswitch_portisch_example.yaml#L261

zd3sf, 2025-02-16:
those are my three 0xB0 commands exactly.

```
cover:
  - platform: time_based
    name: "Living room blinds"
    device_class: blind
    open_action:
      - rf_bridge.send_raw: 
          raw: 'AAB04C0408137702440111139B38192A192A1A1A19292A192A1A19292929292A1A1A1A1A192A19292A1A192A192A1A1A1A1A1A1A1A192A1A1A1A1A1A1A1A1A1A1A1A192A1929292A192A1A1929292955'
    open_duration: 26.26s
    close_action:
      - rf_bridge.send_raw: 
          raw: 'AAB04C0408136E0246010E139B38192A192A1A1A19292A192A1A19292929292A1A1A1A1A192A19292A1A192A192A1A1A1A1A1A1A1A192A1A1A1A1A1A1A1A1A192A1A1A1A192929292A19292929292955'
    close_duration: 25.99s
    stop_action:
      - rf_bridge.send_raw: 
          raw: 'AAB08F0408137502490111139F38192A192A1A1A19292A192A1A19292929292A1A1A1A1A192A19292A1A192A192A1A1A1A1A1A1A1A192A1A1A1A1A1A1A1A1A1A192A1A1A1929292A192929292929295538192A192A1A1A19292A192A1A19292929292A1A1A1A1A192A19292A1A192A192A1A1A1A1A1A1A1A192A1A1A1A1A1A1A1A1A1A192A1A1A1929292A1929292929292955'
    has_built_in_endstop: true
    assumed_state: false
```

mightymos, 2025-02-17:
> Great to hear about the uncovered overflow. Those are my three 0xB0 commands exactly.
> 
> ```

>     close_duration: 25.99s
>     stop_action:
>       - rf_bridge.send_raw: 
> ```

Portisch does actually catch the attempt to write past the end of the buffer.

It seems like for that really long command between 128-160 byte buffer size would be required (length is 0x8F = 143).
Original Portisch had a 112 byte buffer.
Port here has a 96 byte buffer.

I can increase it to 160 bytes on the UU52 dev kit and the attempt to write past buffer end stops being triggered.
I think I am seeing different behavior between microcontrollers that I need to investigate.
I am mainly writing this for my recollection later.

I know it is a funny question but are you certain the stop command is working for your blinds?

zd3sf, 2025-02-17:
Yeah stop command works perfect.

mightymos, 2025-02-23:
> Yeah stop command works perfect.

I'm fairly certain now that the blinds stop command is just repeating data twice.
If you look for the hexadecimal "55" you'll see it in that really long stop string.
0x55 indicates the end of a UART packet.

If you then look at the bytes past 0x55 they are just repeating the data that was already provided.
I checked with an online web tool to compare strings.
I don't know why the sniffing would have produced that output but the 0xB0 command specifies 0x08 repeat transmissions.
The number of repeats could easily just be increased or another rfraw command issued if that is really what is required.

It would be really nice if we knew real world devices did not require really long data packets to work.
Because that is at least one cause of overflows.
We obviously still need to "catch" overflows but it would be better if the example yaml had reasonable examples.

This is the modified stop command which I think is correct - it is the same length as open and close commands:

```
    stop_action:
      - rf_bridge.send_raw: 
          raw: 'AAB04C0408137502490111139F38192A192A1A1A19292A192A1A19292929292A1A1A1A1A192A19292A1A192A192A1A1A1A1A1A1A1A192A1A1A1A1A1A1A1A1A1A192A1A1A1929292A1929292929292955'

```

Could you try it out and tell me if stop still works?
If so, please let me know how it goes.

zd3sf, 2025-02-25:
Hi yes the new stop commands work fine! Thank you for looking into it

mightymos, 2025-03-01:
I fixed at least one source of UART buffer corruption that occurred after running the sniffing mode for long periods of time:
https://github.com/mightymos/RF-Bridge-OB38S003/releases/tag/v0.4.15

I am hoping this improves stability for continuous operation.

zd3sf, 2025-03-01:
Awesome will flash and update this comment with the results

mightymos, 2025-03-02:
> Awesome!! will flash and update this comment with the results,
> 
> Update: 0.4.15 seems to run fine. Bucket send and bucket receive work. Standard transmit/standard receive work great as well. But the beeping sounds strange off tune and choppy. I decided to roll back to 0.4.12 it has been stable for couple of weeks now (with bridge beeping on code receive).

Ok, how strange. What beep duration were you using?

zd3sf, 2025-03-02:
100 ms duration

nicolaerosia, 2025-06-05:
Hello,

sorry for hijacking your issue but I saw you're sending custom commands. I just bought a set of Gaviotta blinders and the remote is on 433MHz. I've found someone who reversed engineered the protocol [1]. Do you think this will work with Sonoff RFBridge v2 using this project?

[1] https://viereck.ch/rf433-gaviota/

nicolaerosia, 2025-06-06:
@zd3sf :) thanks!

Which one is "raw" ?

Took a quick look at the source code, not sure how that will help. Additional protocols seems to disable bucket sniffing (but again, I don't properly understand what bucket sniffing is in the first place) and just enable other protocols and Gaviota is not here.

zd3sf, 2026-09-18:
Just an update on this, I have been running 0.4.12 bucketsniffing firmware  for a bout a year now with no freezes or any problems (manages about 12 sensors, LED strip, and blinds). 

I recently updated to 0.4.16 and the freezing problem returned. I know without logs, we cant chase it down. Just wanted to put this out there if anyone is facing the same problem.