mightymos/RF-Bridge-OB38S003 #20: How to get it working in Tasmota

paulandrewcrouch, 2024-12-08:
I don't know if I'm being really stupid (quite possibly), finally flashed the radio chip with `main_portisch_OB38S003`.

Have flashed the ESP with Tasmota and set it to `Sonoff Bridge (25)`

When I write `RFRaw 177` or 'RFRaw 0x00D' this is all I get

```
21:38:56.048 CMD: rfraw 177
21:38:56.055 MQT: stat/433MHz_Bridge/RESULT = {"RfRaw":"ON"}
21:40:03.728 CMD: RfRaw 0x0D
21:40:03.735 MQT: stat/433MHz_Bridge/RESULT = {"RfRaw":"ON"}
```

Does that mean I've not flashed the radio chip correctly, or should I be doing something different.

Apologies if I've missed the obvious.

mightymos, 2024-12-08:
Hi, please use this release for now:
https://github.com/mightymos/RF-Bridge-OB38S003/releases/tag/v0.4.6

I broke some things in v0.4.7 and have to fix now for the next release.

You should receive an acknowledge when issuing the sniffing command which appears as:
AAA055

Also as far as I know Tasmota cannot understand 0x0D formatted numbers.

Acceptable values are rfraw 177 which is equivalent to rfraw AAB155 (but explicitly shows the start and end hex characters).

paulandrewcrouch, 2024-12-09:
Brillant, working, thank you.

 

https://github.com/mightymos/RF-Bridge-OB38S003/releases/tag/v0.4.6

mightymos, 2024-12-09:
There is a new release.
However, you might check out the other issue thread to see if the testers there have success with that version.

Also, if you are able to post successful decodings/sniffings that helps.
It's primarily just curiosity with how the sonoff is being used.

If not, no problem, and will close comments in a couple weeks since issue was apparently resolved.

paulandrewcrouch, 2024-12-09:
Thanks – it’s reading my car keys, RF remote for the blinds etc – so it’s definitely picking up signals.   What I can’t figure out is the different codes.   Obviously for the car there’s an element of encryption, however for the RF blind I’m a bit surprised.

 

Now I’ve done it, I need to find a use case 😊

 

 

Paul

mightymos, 2024-12-10:
It's possible if you post the blinds myself or someone else could understand the codes.
But generally you're looking for one bit to change.

However, portisch does not support many codes other than standard due to space limits since sniffing is included.
But if you can sniff with 0xB1 you can get an equivalent 0xB0 code with online tools to replay.

closed 2024-12-18 by mightymos