mightymos/RF-Bridge-OB38S003 #21: additional low frequency on send_raw

phineas42, 2024-12-29:
Hi. I'm trying to replay a signal from a 433.92 MHz remote control.
The controlled device does not respond to the replayed signal.
The replayed signal looks different from the original when captured using an RTL-SDR and viewed with Universal Radio Hacker:
In addition to the ASK/OOK pulses, which are replayed with the correct timing, there's a ~ 830 Hz modulation on the high/on pulses.

Disclaimer: I'm new to pretty much all of this, so I could have easily made a silly mistake somewhere. I'm not even completely certain that my device is using ASK/OOK. For all I know, this additional low frequency is standard on these devices.

I'm using RFBridge R2.2, esphome, main_portisch_OB38S003.hex v0.4.9.

I used flashScript.py and recent mightymos/OnbrightFlasher@4917b092d2999737212cb3e0ceee127471c5d914 on an ESP-01s device to flash the radio chip. SDA on GPIO 0 and SCL on GPIO 2 defined in OnbrightFlasher.ino. I also modified the flasher sketch to add multiple write attempts in the event that the return did not match the sent bytes.

mightymos, 2024-12-29:
Okay, we can at least try to look into it.
Can you provide a part number for the remote control and/or the device?
Also please post the portisch output when capturing?
Also you might as well post the URH captures here?

phineas42, 2024-12-31:
Thanks.

Here is the output when capturing the remote button press:
Is this what you mean by "the portisch output"?

```
12:31:10    [D]    [button:010]        'Bucket sniffing start' Pressed.
12:31:10    [I]    [rf_bridge:209]     Raw Bucket Sniffing on
12:31:10    [D]    [uart_debug:114]    >>> AA:B1:55
12:31:10    [D]    [rf_bridge:037]     Action OK
12:31:10    [D]    [uart_debug:114]    <<< AA:A0:55
12:31:12    [I]    [rf_bridge:104]     Received RFBridge Bucket: AA B1 04
3082 57CE 19F9 797C 381A1A 55
12:31:12    [D]    [uart_debug:114]    <<<
AA:B1:04:30:82:57:CE:19:F9:79:7C:38:1A:1A:55
12:31:12    [D]    [uart_debug:114]    >>> AA:A0:55
```

The remote:
MTRF-REM-15NL V1.2
"15 CH Remote 433.92MHz"
FCC ID: VYY-DD5712
Here's an instruction manual for the remote:
https://www.avoutlet.com/images/product/additional/r/mtrf-rem-15-instructions.pdf

I perused the FCC testing docs for the motor device under control. (It is
bi-directional communication)
FCC docs say that Center Frequency is 433.925 MHz, bandwidth is 82.92 KHz.

I captured the spectrum of the remote in operation -- looks like ASK, I
guess.
[image: image.png]

Here's what the captured remote ON pulses look like in URH:
(First 2ms of first pulse)
[image: image.png]

And here's what the replayed remote ON pulses look like in URH:
(First 2 ms of first pulse)
[image: image.png]

Here's a demodulated view of the actual remote button press:
[image: image.png]
(long pulse is ~138 ms, short pulses are ~ 6.74 ms)

And here's the replayed one:
[image: image.png]
(long pulse is ~13 ms, short pulses are ~ 6.83 ms)

I took a look at a few captured samples of that button press on the remote,
and came up with the following RFBridge send stream.
Looking at the bitbucket format, I got the idea that the longest possible
bucket size is 65.535 ms.
The high sync pulse in this protocol appears to be about 138 ms, so I
crafted an RFstream with 3 high pulses to begin.
AA B0 11 04 08 1A4A 7198 9E7A B399 BBB28181818183 55
"BBB" is 3x high bucket 3 (0xB399), or 3 * 45977, which is 137931 us.
Here's the constructed signal, captured, demodulated.
[image: image.png]
(long pulse is ~140 ms)
Again, there is still another waveform shown within the high pulse:
(first 2 ms)
[image: image.png]

phineas42, 2024-12-31:
Thanks.

Here is the output when capturing the remote button press:
Is this what you mean by "the portisch output"?

```
12:31:10    [D]    [button:010]        'Bucket sniffing start' Pressed.    
12:31:10    [I]    [rf_bridge:209]     Raw Bucket Sniffing on        
12:31:10    [D]    [uart_debug:114]    >>> AA:B1:55        
12:31:10    [D]    [rf_bridge:037]     Action OK        
12:31:10    [D]    [uart_debug:114]    <<< AA:A0:55        
12:31:12    [I]    [rf_bridge:104]     Received RFBridge Bucket: AA B1 04 3082 57CE 19F9 797C 381A1A 55    
12:31:12    [D]    [uart_debug:114]    <<< AA:B1:04:30:82:57:CE:19:F9:79:7C:38:1A:1A:55        
12:31:12    [D]    [uart_debug:114]    >>> AA:A0:55
```

The remote:
MTRF-REM-15NL V1.2
"15 CH Remote 433.92MHz"
FCC ID: VYY-DD5712
Here's an instruction manual for the remote:
https://www.avoutlet.com/images/product/additional/r/mtrf-rem-15-instructions.pdf

I perused the FCC testing docs for the motor device under control. (It is bi-directional communication)
FCC docs say that Center Frequency is 433.925 MHz, bandwidth is 82.92 KHz.

I captured the spectrum of the remote in operation -- looks like ASK, I guess.
[image]

Here's what the captured remote ON pulses look like in URH:
(First 2ms of first pulse)
[image]

And here's what the replayed remote ON pulses look like in URH:
(First 2 ms of first pulse)
[image]

Here's a demodulated view of the actual remote button press:
[image]
(long pulse is ~138 ms, short pulses are ~ 6.74 ms)

And here's the replayed one:
[image]
(long pulse is ~13 ms, short pulses are ~ 6.83 ms)

I took a look at a few captured samples of that button press on the remote, and came up with the following RFBridge send stream.
Looking at the bitbucket format, I got the idea that the longest possible bucket size is 65.535 ms.
The high sync pulse in this protocol appears to be about 138 ms, so I crafted an RFstream with 3 high pulses to begin.
AA B0 11 04 08 1A4A 7198 9E7A B399 BBB28181818183 55
"BBB" is 3x high bucket 3 (0xB399), or 3 * 45977, which is 137931 us.
Here's the constructed signal, captured, demodulated.
[image]
(long pulse is ~140 ms)

Again, there is still another waveform shown within the high pulse:
(first 2 ms)
[image]

mightymos, 2024-12-31:
I'm still trying to understand UHR.

But basically it does not look like there are enough data bits here.
For standard formats Portisch should show something like a sync bit, 24 data bits, and an end bit.
Other formats would show more bits, typically multiples of 8 but not less.

I would need to think about what would be most helpful for troubleshooting.

If you're willing to experiment a little you can flash Tasmota because then you can place the microcontroller in sniffing mode and leave it there (otherwise ESPHome sends an ACK which causes sniffing mode to exit).

That way it should be more obvious whether or not you are receiving button presses.
I can post an example output from a remote for outdoor LED lights I have if that would help to understand what to look for.

In any case, if you get more button presses from ESPHome or Tasmota you can post the output here.

mightymos, 2025-01-01:
Here's an example from my door sensor:
```
19:51:09.293 RSL: STATE = {"Time":"2025-01-01T19:51:09","Uptime":"0T00:00:10","UptimeSec":10,"Heap":21,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":30,"MqttCount":0,"Wifi":{"AP":1,"SSId":"Tribble","BSSId":"D8:EC:5E:94:D3:D3","Channel":1,"Mode":"11n","RSSI":100,"Signal":-8,"LinkCount":1,"Downtime":"0T00:00:04"}}
19:51:21.890 CMD: rfraw AAB155
19:51:21.895 RSL: RESULT = {"RfRaw":"ON"}
19:51:21.960 RSL: RESULT = {"Time":"2025-01-01T19:51:21","RfRaw":{"Data":"AAA055"}}
19:51:23.107 RSL: RESULT = {"Time":"2025-01-01T19:51:23","RfRaw":{"Data":"AA B1 03 0190 0384 03A2 281818 55"}}
19:51:24.483 RSL: RESULT = {"Time":"2025-01-01T19:51:24","RfRaw":{"Data":"AA B1 03 0122 03C0 0406 281818 55"}}
19:51:24.634 RSL: RESULT = {"Time":"2025-01-01T19:51:24","RfRaw":{"Data":"AA B1 04 0168 02C6 0078 03AC 381A18 55"}}
19:51:24.656 RSL: RESULT = {"Time":"2025-01-01T19:51:24","RfRaw":{"Data":"AA B1 03 014A 03D4 03C0 281818 55"}}
19:51:25.886 RSL: RESULT = {"Time":"2025-01-01T19:51:25","RfRaw":{"Data":"AA B1 03 017C 03B6 03A2 281818 55"}}
19:51:26.329 RSL: RESULT = {"Time":"2025-01-01T19:51:26","RfRaw":{"Data":"AA B1 03 0186 038E 0370 28181818 55"}}
19:51:27.568 RSL: RESULT = {"Time":"2025-01-01T19:51:27","RfRaw":{"Data":"AA B1 04 07DA 03AC 0186 035C 381A1A1A 55"}}
19:51:27.817 RSL: RESULT = {"Time":"2025-01-01T19:51:27","RfRaw":{"Data":"AA B1 03 017C 0398 0320 281818 55"}}
19:51:28.840 RSL: RESULT = {"Time":"2025-01-01T19:51:28","RfRaw":{"Data":"AA B1 03 0154 03B6 03D4 281818 55"}}
19:51:30.479 RSL: RESULT = {"Time":"2025-01-01T19:51:30","RfRaw":{"Data":"AA B1 03 0172 03B6 2828 28181909081908190819090908181908181818181909090818 55"}}
19:51:31.656 RSL: RESULT = {"Time":"2025-01-01T19:51:31","RfRaw":{"Data":"AA B1 03 019A 0398 0366 281818 55"}}
19:51:33.527 RSL: RESULT = {"Time":"2025-01-01T19:51:33","RfRaw":{"Data":"AA B1 03 019A 0384 03CA 281818 55"}}
19:51:33.942 RSL: RESULT = {"Time":"2025-01-01T19:51:33","RfRaw":{"Data":"AA B1 04 05D2 03C0 015E 03AC 381A1A 55"}}
19:51:34.184 RSL: RESULT = {"Time":"2025-01-01T19:51:34","RfRaw":{"Data":"AA B1 03 0168 0280 03AC 289918 55"}}
19:51:34.452 RSL: RESULT = {"Time":"2025-01-01T19:51:34","RfRaw":{"Data":"AA B1 04 0140 0064 024E 03F2 380928 55"}}
19:51:34.856 RSL: RESULT = {"Time":"2025-01-01T19:51:34","RfRaw":{"Data":"AA B1 03 0172 03AC 03D4 281818 55"}}
19:51:35.291 RSL: RESULT = {"Time":"2025-01-01T19:51:35","RfRaw":{"Data":"AA B1 03 0186 0398 0352 281809 55"}}
19:51:35.448 RSL: RESULT = {"Time":"2025-01-01T19:51:35","RfRaw":{"Data":"AA B1 04 0546 03AC 0190 0384 381A1A 55"}}
19:51:35.939 RSL: RESULT = {"Time":"2025-01-01T19:51:35","RfRaw":{"Data":"AA B1 03 0190 038E 035C 281818 55"}}
19:51:37.184 RSL: RESULT = {"Time":"2025-01-01T19:51:37","RfRaw":{"Data":"AA B1 03 01CC 00B4 03AC 280908 55"}}
19:51:37.416 RSL: RESULT = {"Time":"2025-01-01T19:51:37","RfRaw":{"Data":"AA B1 03 0168 0384 03DE 281818 55"}}
19:51:38.475 RSL: RESULT = {"Time":"2025-01-01T19:51:38","RfRaw":{"Data":"AA B1 04 008C 01D6 03AC 037A 381929 55"}}
19:51:38.590 RSL: RESULT = {"Time":"2025-01-01T19:51:38","RfRaw":{"Data":"AA B1 04 074E 038E 0190 038E 381A1A 55"}}
19:51:39.097 RSL: RESULT = {"Time":"2025-01-01T19:51:39","RfRaw":{"Data":"AA B1 03 019A 0384 03AC 281818 55"}}
19:51:39.150 CMD: rfraw AAA755
19:51:39.155 RSL: RESULT = {"RfRaw":"ON"}
19:51:39.224 RSL: RESULT = {"Time":"2025-01-01T19:51:39","RfRaw":{"Data":"AAA055"}}
19:51:46.538 RSL: RESULT = {"Time":"2025-01-01T19:51:46","RfRaw":{"Data":"AAA4281E015E03F235720A55"}}
19:51:49.287 RSL: RESULT = {"Time":"2025-01-01T19:51:49","RfRaw":{"Data":"AAA42828015403E835720E55"}}
```

You can see most of the sniffed codes are too short.
But there is a longer one that should correspond to the open code for my sensor 0x35720E.
The closed code is 0x35720A.

Here's an analysis I did in the past for a similar sniffed code to understand that it does match:
```
// door sensor
# 313, 955, 10152
# 0x35720E (open)
# = 0b 00110101 01110010 00001110
// 08:39:35.966 RSL: RESULT = {"Time":"2024-11-27T08:39:35","RfRaw":{"Data":"AA B1 03 0139 03BB 27A8          28 18 19 09 08 19 08 19 08              19 09 09 08 18 19 08 18          18 18 18 19 09 09 08 18       55"}}
// AA B0 21 03 08 0139 03BB 27A8      2       81 81 90 90 81 90 81 90     81 90 90 90 81 81 90 81       81 81 81 81 90 90 90 81          8 55

// sync low = 0x2, bit 0 = 0x81, bit 1 = 0x90, sync high = 0x8

// modify to send 0xDEADBE = 0b11011110 10101101 10111110
rfraw AA B0 21 03 08 0138 03B7 277C 29 09 08 19 09 09 09 08 19 08 19 08 19 09 08 19 09 08 19 09 09 09 09 08 18 55
```

phineas42, 2025-01-01:
Sorry for not explicitly mentioning this earlier:
I'm working to control Rollease Acmeda roller shades.

OK, I have flashed Tasmota. Were there any configurations I need to give Tasmota? I chose Module type "Sonoff Bridge (25)", but other than that, I have not modified the defaults.

Here are a few captures:
```19:35:33.247 RSL: RESULT = {"Time":"2025-01-01T19:35:33","RfRaw":{"Data":"AAA055"}}
19:36:05.546 RSL: RESULT = {"Time":"2025-01-01T19:36:05","RfRaw":{"Data":"AA B1 03 19FD 70BE 7115 281818 55"}}
19:36:20.973 RSL: RESULT = {"Time":"2025-01-01T19:36:20","RfRaw":{"Data":"AA B1 05 01D1 2E01 65A5 19FC 7086 48192B2B 55"}}
19:36:25.883 RSL: RESULT = {"Time":"2025-01-01T19:36:25","RfRaw":{"Data":"AA B1 04 30A6 582E 19FB 79F2 381A1A 55"}}
19:37:42.820 RSL: RESULT = {"Time":"2025-01-01T19:37:42","RfRaw":{"Data":"AA B1 04 30A2 58E8 1A02 79DA 381A1A 55"}}
19:37:57.564 RSL: RESULT = {"Time":"2025-01-01T19:37:57","RfRaw":{"Data":"AA B1 08 309F 5862 1A01 0163 00F6 0964 0693 7904 781A1A181A1A181A1A183C5A1A1A1A08181C6A1A181A1A181A1A 55"}}
19:39:36.272 RSL: RESULT = {"Time":"2025-01-01T19:39:36","RfRaw":{"Data":"AA B1 04 30A7 58DC 19FA 7A38 381A1A 55"}}
19:39:48.741 RSL: RESULT = {"Time":"2025-01-01T19:39:48","RfRaw":{"Data":"AA B1 05 309D 1D21 13D5 739F 7976 48182828282828283939 55"}}```

Now I've done a bit more forum perusal, and I'm not the only one having issues with this remote/device.
In fact, [this post for he Bond bridge](https://forum.bondhome.io/t/rollease-acmeda-and-dooya-motorized-shades/413/19) looks a lot like my previous note here on this issue, with similar screenshots of the strange modulation. Bond further indicates that GFSK is used by my remote/device.

Because of that other post showing the same differences in the reproduction of the analog signal, I'm guessing that there's not a bug with any of esphome/tasmota/your OBS38S003 firmwares, other than possibly eventual feature requests for this protocol.

mightymos, 2025-01-01:
You can just use module type Sonoff Bridge (25).
If the protocol is really GFSK, that's completely different from simple on-off keying.

I see a code in your output that looks to be 24-bits of data.
You can use this tool to obtain the corresponding transmit code (0xB0):
https://bbconv.hrbl.pl/

This is what I get from your sniffed code (0xB1) using the tool:
AA B0 2C 08 08 309F 5862 1A01 0163 00F6 0964 0693 7904 781A1A181A1A181A1A183C5A1A1A1A08181C6A1A181A1A181A1A 55

You would put this in Tasmota to try transmitting:
```
rfraw AA B0 2C 08 08 309F 5862 1A01 0163 00F6 0964 0693 7904 781A1A181A1A181A1A183C5A1A1A1A08181C6A1A181A1A181A1A 55
```

phineas42, 2025-01-03:
Thanks for the help. Putting that code into tasmota did not work.

My devices are definitely a type of FSK, and now that I know better, I've found various documentation that corroborates this. I think it's best to close this as a non-issue.

Thanks again.

closed 2025-01-03 by phineas42