mightymos/RF-Bridge-OB38S003 #22: Having issues with ESPhome but not Tasmota...

AaronAnz, 2025-01-04:
Firstly, awesome work and great write up. I'm new to esp devices realm and it was pretty easy to follow but i'm having a weird issue with esphome. I prefer the look and configuration methods for esphome over tasmota so I really wanna get it to work in favor of Tasmota but im having issues. 

I have a Sonoff RF Bridge R2.2. I successfully flashed the Passthrough firmware you provided flashed to the RF chip and when I flash the ESP chip with Tasmota and configured as you designated and everything works great with this:
```
RfSend 0xA110E0E, 32, 1, 5, 269
RfSend 0xA111E1E, 32, 1, 5, 270
```

but when I flash esphome this wont work to turn to control my device:

```
substitutions:
  name: "sonoff-rf-bridge-r2"
  friendly_name: "Sonoff RF Bridge r2.2"

esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  comment: Sonoff RF BridgeR2 with ESPHome

esp8266:
  board: esp8285

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  fast_connect: True
  power_save_mode: NONE
  manual_ip: 
    static_ip: !secret device_ip
    gateway: !secret gateway
    subnet: !secret subnet
    dns1: !secret gateway
  ap:
    ssid: "${friendly_name} Backup AP"
    password: !secret wifi_password

web_server:
  port: 80
  ota: true
  local: true

ota:
  - platform: esphome
    password: !secret ota_password

logger:
  level: DEBUG
  hardware_uart: UART1

captive_portal:

api:
  password: !secret api_password

status_led:
  pin: GPIO13

remote_receiver:
  pin:
    number: GPIO3
    mode:
      input: true
      pullup: false
      
  tolerance: 50%
  filter: 4us
  idle: 4ms
  dump: rc_switch

remote_transmitter:
  pin: GPIO1
  carrier_duty_percent: 100%

button:
  - platform: template
    name: "Send RF Code OFF"
    on_press:
      - remote_transmitter.transmit_rc_switch_raw:
          code: "00001010000100010000111000001110" # 0x0xA110E0E padded to 32 bits
          protocol: 1
          repeat: 5
```

Any help would be greatly appreciated!

mightymos, 2025-01-04:
I do not know yet if this will fix it but Protocol 1 has a base pulse length of 350 microseconds apparently which you can see in the table here for original RCSwitch:
https://github.com/sui77/rc-switch/blob/master/RCSwitch.cpp

However, your Tasmota command that was working used 270 pulse length.
So I guess you need to additionally specify the pulse length in ESPHome like is explained here to be 270:
https://esphome.io/components/remote_transmitter.html#remote-transmitter-rc-switch-protocol

AaronAnz, 2025-01-04:
So I guess then ESPHome doesn't differentiate any pulse_length when receiving cause it comes back as this in esphome:
```
23:01:24 | [I] | [remote.rc_switch:261] | Received RCSwitch Raw: protocol=1 data='00001010000100010001111000001111'
23:01:24 | [I] | [remote.rc_switch:261] | Received RCSwitch Raw: protocol=1 data='00001010000100010001111000001111'
23:01:24 | [I] | [remote.rc_switch:261] | Received RCSwitch Raw: protocol=1 data='00001010000100010001111000001111'
23:01:24 | [I] | [remote.rc_switch:261] | Received RCSwitch Raw: protocol=1 data='00001010000100010001111000001111'
23:01:24 | [I] | [remote.rc_switch:261] | Received RCSwitch Raw: protocol=1 data='00001010000100010001111000001111'
```
I had found part of the documentation and for some reason I had tried just adding the pulse_length property to the transmitter itself ( which obviously gave an error ) but for some reason I didn't add it to the protocol section itself.... I guess me being a noob with RF was part of the problem. 

Anyways, so it still doesn't seem to work even by setting that. Do you know how to get esphome to output better details on the protocol?

the-mentor, 2025-01-04:
Hi @AaronAnz I highly recommend that you use an external device to decode the signal since the raw decoding on esphome is very bad

I decoded all my signals using universal radio hacker and a USB RTL-SDR dongle

I also wrote a website that converts from the universal radio hacker hex to esphome raw

https://the-mentor.github.io/Convert-HexToESPHomeRaw/

AaronAnz, 2025-01-04:
> Hi @AaronAnz I highly recommend that you use an external device to decode the signal since the raw decoding on esphome is very bad
>
> I decoded all my signals using universal radio hacker and a USB [...]

I considered that but didn't really want to buy more stuff for this project. I dont have enough RF devices to warrant spending more time and money on this lol. Tasmota it is for now, maybe I will revisit in the future. 

Thanks both of you for your responses.

mightymos, 2025-01-04:
@AaronAnz Avri has been helping people with decoding. One last thing you could do only with your existing hardware is to flash the Portisch port to the microcontroller. Then use sniffing mode to attempt decoding with rfraw AA B1 55 in Tasmota.

You can post the output from Tasmota here during sniffing and I could attempt to tell if it seems legitimate or not.
I would assume it would appear similar to the data you posted but would hopefully contain more timing information that might be helpful. If that goes well you might also be able to transmit using Portisch as well.

AaronAnz, 2025-01-04:
> @AaronAnz Avri has been helping people with decoding. One last thing you could do only with your existing hardware is to flash the Portisch port to the microcontroller. Then use sniffing mode to [...]

```
09:48:10.791 RSL: RESULT = {"Time":"2025-01-04T09:48:10","RfRaw":{"Data":"AA B1 03 012A 0361 205A 281818181908190818181819081818190818181819090908181818190909090908 55"}}
09:48:13.049 RSL: RESULT = {"Time":"2025-01-04T09:48:13","RfRaw":{"Data":"AA B1 03 012B 0363 2058 281818181908190818181819081818190818181909090908181819081909081908 55"}}
09:48:15.425 RSL: RESULT = {"Time":"2025-01-04T09:48:15","RfRaw":{"Data":"AA B1 03 012C 0361 2058 281818181908190818181819081818190818181819090908181819090909081908 55"}}
09:48:17.616 RSL: RESULT = {"Time":"2025-01-04T09:48:17","RfRaw":{"Data":"AA B1 03 012D 0360 2058 281818181908190818181819081818190818181909090908181908181908190908 55"}}
09:48:20.016 RSL: RESULT = {"Time":"2025-01-04T09:48:20","RfRaw":{"Data":"AA B1 03 0129 0363 205A 281818181908190818181819081818190818181819090908181908190908190908 55"}}
09:48:22.467 RSL: RESULT = {"Time":"2025-01-04T09:48:22","RfRaw":{"Data":"AA B1 03 012B 0362 2057 281818181908190818181819081818190818181909090908181909081908181908 55"}}
```

mightymos, 2025-01-04:
@AaronAnz 

Ok, I do not understand about the changing codes unless it's a rolling code thing (is it a garage door remote?).
I realize you're trying to recover the actual data rather than a sniffed code, but I'd like to see if we can get anything to work as a start.

Those codes seem to be the standard timings but instead of 24 bits have 32 bits instead as you have said.

So you can get a code to replay (0xB0) using this tool:
https://bbconv.hrbl.pl/

And use the rf_bridge component in ESPHome to send as shown in the other example yaml.
So rf_bridge.send_raw for (0xB0) command and you can try rf_bridge.send_advanced_code since you know the data.
But anyway, try send_raw first.

I would like to check that you decoded data correctly once I have time.
I will also check out the passthrough and just confirm it remains working on my hardware.

AaronAnz, 2025-01-04:
@mightymos 

Sorry if I worded that poorly.  That conversion website was successful for me. 

```
# original which doesnt transmit correctly 
AA B1 03 012A 0361 205A 281818181908190818181819081818190818181819090908181818190909090908 55"}

# converted with the website you provided does transmit correctly
AA B0 29 03 08 012A 0361 205A 281818181908190818181819081818190818181819090908181818190909090908 55
```

So now that I know you need to decode the captured data with that website I should be able to get it working with ESPHome now. You're awesome! Thank you so much!

I guess last question would be is there a way to force decoding of the rf data automatically that comes in to make my data gathering easier or is using that website just the easiest?

I don't think this matters anymore but in-case you're curious for a better explanation of the remote:

its a remote for an LED lamp that has a single button for toggling on and off the lamp. It goes as follows when you press it consecutively :
```
# tasmota + passthrough
ON: RfSend 0xA111E1E, 32, 1, 5, 270
OFF: RfSend 0xA110E0E, 32, 1, 5, 270
ON: RfSend 0xA111E1E, 32, 1, 5, 270
OFF: RfSend 0xA110E2C, 32, 1, 5, 270
ON: RfSend 0xA111E3C, 32, 1, 5, 270
OFF: RfSend 0xA110E4A, 32, 1, 5, 270
ON: RfSend 0xA111E5A, 32, 1, 5, 270
OFF: RfSend 0xA110E68, 32, 1, 5, 270
ON: RfSend 0xA111E78, 32, 1, 5, 270
OFF: RfSend 0xA110E86, 32, 1, 5, 270
ON: RfSend 0xA111E96, 32, 1, 5, 270
OFF: RfSend 0xA110EA4, 32, 1, 5, 270
ON: RfSend 0xA111EB4, 32, 1, 5, 270
OFF: RfSend 0xA110EC2, 32, 1, 5, 270
ON: RfSend 0xA111ED2, 32, 1, 5, 270
OFF: RfSend 0xA110EE0, 32, 1, 5, 269
ON: RfSend 0xA111EF0, 32, 1, 5, 270
OFF: RfSend 0xA110EFE, 32, 1, 5, 269
```
Then it starts from the top. 

So on range is 0xA111E1E - 0xA111EF0 incrementing by 0x1E
and off range is 0xA110E0E - 0xA110EFE incrementing by 0x1E

mightymos, 2025-01-04:
You could manually convert 0xB1 command to 0xB0 fairly easily if you wanted to.
You'll notice the data portion is exactly the same.

Then 0x29 = 41 is the packet length, 0x03 is the number of timing buckets, and 0x08 is the number of repeats.
The format is defined here:
https://github.com/Portisch/RF-Bridge-EFM8BB1/wiki/0xB0

As far as the LED light, it's interesting that it increments in some way.
I'll have to think about it more to understand what might be going on.

But glad the bridge is working successfully now.

mightymos, 2025-01-20:
Closed as working.