mightymos/RF-Bridge-OB38S003 #23: Feature suggestion: Expose buzzer

justinkb, 2025-01-15:
I was thinking it would be cool if we could have the firmware "expose" the buzzer by mirroring some signal coming from the esp8285, in the way the passthrough firmware mirrors the radio signals on the OB38S003 in the other direction. Then I could have home assistant tell esphome to tell the onbright to buzz on receiving a specific signal. Specifically looking to "mirror" my doorbell on the sonoff which is one storey higher (the doorbell can be hard to hear there). I actually wanted to try to implement this myself, but I have the sonoff operational now and didn't really want to open it up right now. And I couldn't figure out which pins on either side are actually available as means of communicating between the two MCUs on the board that aren't already used for something else.

mightymos, 2025-01-15:
Why not just use the beep action?
https://esphome.io/components/rf_bridge.html#rf-bridge-beep-action

I think it would be much more straightforward?

justinkb, 2025-01-16:
> Why not just use the beep action? https://esphome.io/components/rf_bridge.html#rf-bridge-beep-action
> 
> I think it would be much more straightforward?

I did not know that existed. Thanks for pointing that out. Does it work on the onbridge variant too?

mightymos, 2025-01-16:
Yes that's true, it's only supported in portisch.
It would be straightforward to add to rcswitch as well.

As I recall, they did not break out any other GPIO into connectable pads except for the RESET pin on the microcontroller.

justinkb, 2025-01-16:
That's a shame. Could I use that reset pin? Or would that run into issues? If so I'm not scared of soldering onto a really tiny pin directly in the chip, but if it is avoidable, that's obviously better

mightymos, 2025-01-16:
I ultimately used the Busy Bee kit if I needed extra pins to toggle for debugging.
The fuse bytes setting the function of the pin as GPIO would need to be changed, and it might require a pull up resistor.

Ideally I think you'd just use the Portish port.
We can try to predict if it is compatible, what devices do you need it to work with?

Another alternative that I still think would be much simpler is to use something like the ESPToy v2 with an onboard buzzer:
https://opensprinkler.com/product/esptoy/
https://github.com/mightymos/esptoy_v200

You could trigger the buzzer from Home Assistant and even have it play little tunes.

justinkb, 2025-01-16:
I chose the passthrough because it seemed the simplest way to go and I didn't know if the portisch port for onbright was mature enough, probably will just switch to it. Have no hard requirement for passthrough, I'll let you know if I run into any issue

zd3sf, 2025-01-20:
I think it worked on the RCswitch too when I was using it.

mightymos, 2025-01-20:
@zd3sf 
Ok, I think I just fixed the beep duration for rcswitch.
It was the same issue as in Portisch, SDCC for 8051 is little endian but data is sent as big endian.
Did not test.

@justinkb 
Closing issue for now, but post again to report success or issues.

closed 2025-01-20 by mightymos