mightymos/RF-Bridge-OB38S003 #24: I don't receive "standard" codes in advanced sniffing mode

koying, 2025-01-17:
I don't actually know if it's a bug or by design, but I don't receive "standard" EV1527 / 24 bits codes in advanced sniffing mode.
The red led doesn't lit up at all in sniffing mode, if that matters.

All fine in "standard" mode.

Using esphome, if that matters, but I enabled uart debugging, and I don't see anything on the lline, either.

mightymos, 2025-01-17:
So I have been putting off looking into this lol.

So on original Portisch (Sonoff black box) the advanced decoding does not appear to work either.

This is the output I get for my door sensor:
```
23:26:07.344 QPC: Reset
23:26:09.263 RSL: STATE = {"Time":"2025-01-17T23:26:09","Uptime":"0T00:00:09","UptimeSec":9,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"Tribble","BSSId":"D8:EC:5E:94:D3:D3","Channel":1,"Mode":"11n","RSSI":100,"Signal":-10,"LinkCount":1,"Downtime":"0T00:00:04"}}
23:26:52.465 RSL: RESULT = {"Time":"2025-01-17T23:26:52","RfReceived":{"Sync":10270,"Low":340,"High":1000,"Data":"35720E","RfKey":"None"}}
23:27:00.514 CMD: rfraw AA A6 55
23:27:00.519 RSL: RESULT = {"RfRaw":"ON"}
23:27:00.583 RSL: RESULT = {"Time":"2025-01-17T23:27:00","RfRaw":{"Data":"AAA055"}}
```

Even after triggering the door sensor several more times there is no more decodings.
Once I stop advanced sniffing with 0xA7 it goes back to decoding.

I will try to look into this once there is time.

koying, 2025-01-18:
Something else I noticed is that in "bucket sniffing", I receive a single output, then the firmware goes immediately back to "standard". 

According to posts I've read, bucket mode is supposed to stick, doesn't it?

zd3sf, 2025-01-19:
Can confirm that I have noticed both issues, bucket sniffing goes back to standard and and I dont remember that I was able to get anything on advanced sniffing.

koying, 2025-01-20:
PR #26 revert to what original portisch does regarding buckets, really.
Ignoring whitespaces, it's a one-liner: https://github.com/mightymos/RF-Bridge-OB38S003/pull/26/files?diff=unified&w=1#diff-5284d85a5d01e5105be2e42ce9b426e8516bd1dfc33390d3ef26f2f8ae737672R207

closed 2025-01-20 by mightymos

mightymos, 2025-01-20:
> PR [#26](https://github.com/mightymos/RF-Bridge-OB38S003/pull/26) revert to what original portisch does regarding buckets, really. Ignoring whitespaces, it's a one-liner: [...]

Ahh okay, thanks for the catch.

zd3sf, 2025-01-20:
This has solved the bucket sniffing issue. I still don't see anything on advanced sniffing. Not sure if we need a specific kind of remote to emit those advanced signals or should a standard transmit also show up in advanced?

mightymos, 2025-01-20:
> This has solved the bucket sniffing issue. I still don't see anything on advanced sniffing. Not sure if we need a specific kind of remote to emit those advanced signals or should a standard transmit also show up in advanced?

The standard decoding measures the sync length (e.g., 10850 microseconds).
Then the other timings are calculated as fractions of that measurement.
So dividing by 31 is 350 microseconds and multiplying that by 3 is 1050 microseconds.
Finally, data bits are compared to these measured timings to look for a match.

Advanced decoding instead compares to a look up table.
However the only advanced protocol we can fit in code space now is the standard:
```
#if EFM8BB1_SUPPORT_PT226X_PROTOCOL == 1
#define PT226X
__code static uint16_t PROTOCOL_BUCKETS(PT226X)[] = { 350, 1050, 10850 };
__code static uint8_t  PROTOCOL_START(PT226X)[]   = { HIGH(0), LOW(2) };
__code static uint8_t  PROTOCOL_BIT0(PT226X)[]    = { HIGH(0), LOW(1) };
__code static uint8_t  PROTOCOL_BIT1(PT226X)[]    = { HIGH(1), LOW(0) };
#endif

So in either mode the same standard protocol SHOULD be decoded.

Original Portisch fit in more protocols because I think the Keil compiler optimized better.
Also it was possible to exclude sniffing in exchange for inclusion of more protocol timings.

To start I am going to confirm advanced decoding behavior of original Portisch on my original black box Sonoff.

mightymos, 2025-01-20:
@zd3sf 
The advanced decoding is happening successfully but not making it to wifi devices.
I do not yet know why.

These are the bytes that are output for open and then close of door sensor:
AA A6 04 00 35 72 0A 55 AA A6 04 00 35 72 0E 55  

I do not know if reported protocol byte 00 should maybe be 01 or not.

koying, 2025-01-21:
> However the only advanced protocol we can fit in code space now is the standard:

What I did to fit an additional protocol in was to comment out all code regarding transmission, that I don't need.
Might be an idea to add a build option...

PR #26 (2025-01-20): Stay in bucket mode
Fixes #24 / 2

There are a bunch of EOL changes that can be ignored
PR #25 (2025-01-20): FIX: bool cast
Never try to outsmart the compiler 😂

Fixes #24 

There are a bunch of EOL changes that can be ignored