mightymos/RF-Bridge-OB38S003 #27: Timings for bucket sending are off

chaosmaster, 2025-01-30:
Hi, first of all, thank you for your great work in porting this to SDCC and the OB38S003. I did a similar project with an STM8-based tuya-mcu and SDCC in the past, so I know the pain with size-constraints.

I am seing the following issue, where it seems when sending the B0 commands the timings are not quite right.

Data sniffed from original remote:
`{"RfRaw":{"Data":"AA B1 03 0130 036D 2444 28181819081909090909090908181908181818190908181818 55"}}`

Converted to B0 (This is not picked up by the receiver):
`AA B0 21 03 08 0130 036D 2444 28181819081909090909090908181908181818190908181818 55`

I used a different device to capture and replay what is sent by the bridge after sending above command and sniffed that again with B1:
`{"RfRaw":{"Data":"AA B1 03 016A 039B 24A4 28181819081909090909090908181908181818190908181818 55"}}`

As you can see the timings are different from the command sent by the original remote (about 30 more for each bucket)

I modified the above B0 command by subtracting 30 from each of the bucket timings and this time the command is picked up by the receiver correctly:
`AA B0 21 03 08 0100 033D 2414 28181819081909090909090908181908181818190908181818 55`

For completeness I again captured and replayed it, to sniff it again with B1 and you can see the timings are now much closer to the original:
`{"RfRaw":{"Data":"AA B1 03 0134 036E 2461 28181819081909090909090908181908181818190908181818 55"}}`

chaosmaster, 2025-02-07:
I just realized this is actually a standard 24-bit command.
While the default "Sonoff"-Mode allows learning this command, the issue is the same here, when trying send the learned command the timings are off and I have to adjust them accordingly.

Learned command:
`{"Sync":9315,"Low":326,"High":895,"Data":"17F218"}`

Adjusted command:
`{"Sync":9200,"Low":200,"High":800,"Data":"17F218"}`

mightymos, 2025-02-07:
I am just confused because the working times that have been provided for low for example are equivalent to 200, 256, and 308 microseconds.

That's actually a fairly wide range.

And the ratio between low and high should be about 3 for the so called standard timings.
Then the ratio of low to sync should be about 31.

But in any case I don't know what the true value is expected to be in this case.

You mentioned measuring with a different device, what is that device?
Oscilloscope?

At various points in time I measured timings with an oscilloscope.
I can try to check recent firmwares again the same way once there is time.

chaosmaster, 2025-02-07:
Unfortunately I don't have easy access to an oscilloscope at the moment.
I used a Flipper Zero to capture and replay the signal.

For completeness here is the adjusted command above replayed through the Flipper:

`{"Sync":9276,"Low":295,"High":828,"Data":"17F218"}`

the unadjusted command is not even picked up in this mode when replayed thorugh the Flipper.

It seems to me the measured timings during recording are correct, but using the same timings for sending, there is some delay added that results the actual timings on air becoming too large.

If there is anything else I can test let me know.

phoenix-9, 2025-10-18:
I might be having the same issue.  I use the bucket sniffing binary.  I set the timings at 400us and 796us for the main buckets:
AAB058040301901440031C271480808...

When I send the code with rf_bridge.send_raw I can measure the timings with an SDR dongle and rtl_433  on my computer.  I get 476us and 872us.

I went through the trouble of defining the new protocol (Hunter ceiling fans) in portisch_protocols.h and recompiled the firmware.  I used 400us and 800us for the buckets.  If I send the code with rf_bridge.send_advanced_code and measure timings with the SDR dongle I get 476us and 880us.

Still, the question is:  Who is right?  Is my SDR dongle off by 20% or is the OB38S003 firmware off?  It really does not matter.  If you have the original remote, you can adjust the bucket timings so that the code sent by the OB38S003 matches the code from the original receiver.

If I had a scope or logic analyzer that I know is accurate and precise, I would measure the timings and look into adjusting the code.

mightymos, 2025-10-19:
Yeah original Portisch had an adjustment for timer startup delay that seems to be experimentally determined.
In other words, I did not see a value provided in the datasheet.

I would like the timing to be accurate.
However I no longer have this hardware.

It would probably be less expensive to use a USB logic analyzer with Pulseview for examining timings.
Also I found that working with one of the fancier EFM8 development boards with a generic 433 MHz receiver attached provided more debugging options (e.g., 2 x UART and additional pins to toggle).

It would be nice if Onbright had sold development boards but did not find any listings.

mightymos, 2025-10-25:
I remembered this person got a protocol partially working with 4 sync bits:
https://github.com/mightymos/RF-Bridge-OB38S003/issues/31

It seems like a lot of work to add a protocol.

zd3sf, 2026-08-20:
Hi mightymos, I have tried this release 2 days ago. I couldn't send Bitbucket commands or receive standard commands. So, I reverted to the latest 0.4.x. If you are in the US, I have a spare bridge I can send to you if you want.

mightymos, 2026-08-20:
> Hi mightymos, I have tried this release 2 days ago. I couldn't send Bitbucket commands or receive standard commands. So, I reverted to the latest 0.4.x. If you are in the US, I have a spare bridge I can send to you if you want.

My "fix" also included a basic mistake, which I think I've fixed.
Could you please try this version instead?

Sorry for the trouble.
If that does not work either I will have to consider obtaining another sonoff hardware.

[passthrough_main_OB38S003.zip](https://github.com/user-attachments/files/31280149/passthrough_main_OB38S003.zip)

zd3sf, 2026-08-20:
Sorry I meant to say that I am on bucket sniffing firmware

mightymos, 2026-08-20:
> Sorry I meant to say that I am on bucket sniffing firmware

If you download the zip it should also contain the sniffing firmware.
The zip file is just poorly named.

zd3sf, 2026-08-21:
Got it! I tested standard send and standard receive, they seem to work. Bucket sniffing works and the output seems to be consistent (no more chopped up bucket sniffs). B1 send works (from codes captured on 0.4.9 fw). Bucket B1 send doesn't seem to do anything on my shades using B0 captured on this firmware.