mightymos/RF-Bridge-OB38S003 #29: Thanks for sharing the Sources and Flasher

rtek1000, 2025-02-05:
Hello,

Thanks for sharing the sources and [fasher](https://github.com/rtek1000/OnbrightFlasher)files, I will try to compile a program for a PAN TILT support for camera that came with the OB38S003.

Unfortunately it came with a speed of 2400 baud, but the camera only sends 9600 baud.

I tried to make an interface to convert 9600 to 2400 but because the 2400 side takes longer, it ends up running over the data, making it necessary to use FIFO which seems more complicated to me than making a new program.

[image]

mightymos, 2025-02-05:
That's an interesting problem.

It does seem like a replacement firmware might be the easier approach ultimately.
Is the microcontroller locked?
Are you able to dump the stock firmware?

rtek1000, 2025-02-05:
I still don't know how the microcontroller is, the board works. I'll upload the board schematic, but it's simple, it has the microcontroller, two unipolar stepper motor drivers.

But it's been a while since I used SDCC, and when I did it was for STM8S003. I also used the Eclipse IDE, in this project for the [W1209](https://github.com/rtek1000/W1209-firmware-modified/tree/master/W1209/W1209-firmware-Eclipse/w1209-firmware-modified-eclipse) thermostat board.

I changed the computer, so I'm installing the tools again, I tried to install SDCC on Windows 11, but I didn't make much progress, when the 'make' command was missing. I'm trying now on Linux.

rtek1000, 2025-02-05:
When I ran make, it gave an error about dos2unix:

> Compiling src/main_passthrough.c
> mkdir -p object/
> sdcc -mmcs51 --model-small  -DTARGET_BOARD_OB38S003 -DBUCKET_SNIFFING_INCLUDED -DMCU_FREQ=16000000UL -Iinc/ -Idrivers/ob38s003/inc -c -o object/main_passthrough.rel src/main_passthrough.c
> Compiling drivers/ob38s003/src/delay.c
> mkdir -p object/
> sdcc -mmcs51 --model-small  -DTARGET_BOARD_OB38S003 -DBUCKET_SNIFFING_INCLUDED -DMCU_FREQ=16000000UL -Iinc/ -Idrivers/ob38s003/inc -c -o object/delay.rel drivers/ob38s003/src/delay.c
> Compiling drivers/ob38s003/src/hal.c
> mkdir -p object/
> sdcc -mmcs51 --model-small  -DTARGET_BOARD_OB38S003 -DBUCKET_SNIFFING_INCLUDED -DMCU_FREQ=16000000UL -Iinc/ -Idrivers/ob38s003/inc -c -o object/hal.rel drivers/ob38s003/src/hal.c
> Linking object/main_passthrough.rel object/delay.rel object/hal.rel
> mkdir -p build/
> sdcc -mmcs51 --model-small --iram-size 256 --xram-size 256 --code-size 8192 -o build/main_passthrough_OB38S003.ihx object/main_passthrough.rel object/delay.rel object/hal.rel
> # hex lines are a short, fixed length (compared with ihx) and therefore works with upload tools
> packihx build/main_passthrough_OB38S003.ihx > build/main_passthrough_OB38S003.hex
> packihx: read 22 lines, wrote 33: OK.
> # unix style line endings (LF instead of LFCR) work with upload tools
> dos2unix build/main_passthrough_OB38S003.hex
> make: dos2unix: No such file or directory
> make: *** [Makefile:213: build/main_passthrough_OB38S003.ihx] Error 127
>

rtek1000, 2025-02-05:
I found how to install dos2unix, but it seems there was an error in the source code:

`sudo apt-get install dos2unix`

When I ran make:
```
Compiling src/main_portisch.c
mkdir -p object/
sdcc -mmcs51 --model-small  -DTARGET_BOARD_OB38S003 -DBUCKET_SNIFFING_INCLUDED -DMCU_FREQ=16000000UL -Iinc/ -Idrivers/ob38s003/inc -c -o object/main_portisch.rel src/main_portisch.c
src/main_portisch.c:417: syntax error: token -> 'uint8_t' ; column 27
src/main_portisch.c:428: error 20: Undefined identifier 'num_buckets'
src/main_portisch.c:432: error 20: Undefined identifier 'num_buckets'
make: *** [Makefile:235: object/main_portisch.rel] Error 1
```

rtek1000, 2025-02-05:
Please, what version of SDCC was used for this RF-Bridge-OB38S003 program?

mightymos, 2025-02-06:
I think I am on sdcc version 4.4.0, though v4.5.0 was recently released.

Ii looks like stdint.h just needs to be included.
I am not sure why it would work on Windows but not on Linux.
I can double check later.

Well, I would think you need to have a schematic first to see what flashing entry points the board supports.
On the Sonoff hardware the reset pin was originally disabled.
So to enter flashing mode required handshake over I2C at startup.
But the fuse options can be set by the programmer to anything desired.

rtek1000, 2025-02-06:
This is the partial schematic, only two connectors not placed on the board need to be connected, and some small bypass capacitors.

So many pins left, they used the RESET Pin for signal output.

[image]

mightymos, 2025-02-06:
Ok, have you seen the OB38S003 datasheet:
https://github.com/arendst/Tasmota/files/9624969/ISSFD-M123_A_OB38S003-Datasheet.EN.-20171106.pdf

For in circuit programming with an external tool (without bootloader) you will need P1.2 and P1.3.
Are those brought out to a connector?

rtek1000, 2025-02-06:
> P1.2 and P1.3. Are those brought out to a connector?

These pins are on connector J2.

mightymos, 2025-02-06:
> > P1.2 and P1.3. Are those brought out to a connector?
> 
> These pins are on connector JP2.

Ok that is helpful.
JP2 must be the programming header.

rtek1000, 2025-02-06:
> Ii looks like stdint.h just needs to be included. I am not sure why it would work on Windows but not on Linux. I can double check later.

The stdint.h was already included,

By declaring the variables at the beginning of the function, the error stops happening:

```
bool radio_state_machine(const uart_command_t command)
{
	bool completed = false;

	// DEBUG:
	//const uint16_t pulsewidths_dummy[3] = {350, 1050, 10850};
	//const uint8_t  packet_dummy[3]  = {0xA5, 0x5A, 0xA5};

    // for transmission
    uint16_t pulsewidths[3];
    uint8_t  tr_packet[3];

	// helps allow sendbuckets call to be more readable
	uint8_t start_size;
	uint8_t bit0_size;
	uint8_t bit1_size;
	uint8_t end_size;
	uint8_t bitcount;

	uint8_t num_buckets;

	uint8_t* rfdata;
	uint8_t data_len;
    	uint16_t* buckets_pointer;
```

Now he says he doesn't have enough resources:
(Insufficient ROM/EPROM/FLASH memory)

```
Compiling src/main_portisch.c
mkdir -p object/
sdcc -mmcs51 --model-small  -DTARGET_BOARD_OB38S003 -DBUCKET_SNIFFING_INCLUDED -DMCU_FREQ=16000000UL -Iinc/ -Idrivers/ob38s003/inc -c -o object/main_portisch.rel src/main_portisch.c
Compiling src/portisch.c
mkdir -p object/
sdcc -mmcs51 --model-small  -DTARGET_BOARD_OB38S003 -DBUCKET_SNIFFING_INCLUDED -DMCU_FREQ=16000000UL -Iinc/ -Idrivers/ob38s003/inc -c -o object/portisch.rel src/portisch.c
Compiling src/portisch_serial.c
mkdir -p object/
sdcc -mmcs51 --model-small  -DTARGET_BOARD_OB38S003 -DBUCKET_SNIFFING_INCLUDED -DMCU_FREQ=16000000UL -Iinc/ -Idrivers/ob38s003/inc -c -o object/portisch_serial.rel src/portisch_serial.c
Compiling drivers/ob38s003/src/timer_interrupts.c
mkdir -p object/
sdcc -mmcs51 --model-small  -DTARGET_BOARD_OB38S003 -DBUCKET_SNIFFING_INCLUDED -DMCU_FREQ=16000000UL -Iinc/ -Idrivers/ob38s003/inc -c -o object/timer_interrupts.rel drivers/ob38s003/src/timer_interrupts.c
Compiling src/uart.c
mkdir -p object/
sdcc -mmcs51 --model-small  -DTARGET_BOARD_OB38S003 -DBUCKET_SNIFFING_INCLUDED -DMCU_FREQ=16000000UL -Iinc/ -Idrivers/ob38s003/inc -c -o object/uart.rel src/uart.c
Linking object/main_portisch.rel object/portisch.rel object/portisch_serial.rel object/timer_interrupts.rel object/uart.rel object/hal.rel
mkdir -p build/
sdcc -mmcs51 --model-small --iram-size 256 --xram-size 256 --code-size 8192 -o build/main_portisch_OB38S003_BUCKET_SNIFFING_INCLUDED.ihx object/main_portisch.rel object/portisch.rel object/portisch_serial.rel object/timer_interrupts.rel object/uart.rel object/hal.rel

?ASlink-Error-Insufficient ROM/EPROM/FLASH memory.
make: *** [Makefile:226: build/main_portisch_OB38S003_BUCKET_SNIFFING_INCLUDED.ihx] Error 1
```

rtek1000, 2025-02-06:
I will try to port the project to the Eclipse IDE, to make it easier to navigate between functions.

rtek1000, 2025-02-07:
> SDCC v4.2.0 (from Debian apt) and v4.5.0 (on Windows) apparently behave differently. As you say, declaring the variables at the function beginning does fix the compilation error for both.
> 
> However, program fits into flash space with v4.5.0.

I left a note there in the SDCC community:
https://sourceforge.net/p/sdcc/bugs/3832/

mightymos, 2025-02-07:
> I left a note there in the SDCC community: https://sourceforge.net/p/sdcc/bugs/3832/

Alright, I mean the v4.2.0 is almost three years old.
So they apparently generate more compact code with v4.5.0 so there's been an improvement.

Debian (testing) is up to v4.4.0 so that would also be suitable:
https://packages.debian.org/search?keywords=sdcc

rtek1000, 2025-02-07:
> Alright, I mean the v4.2.0 is almost three years old. So they apparently generate more compact code with v4.5.0 so there's been an improvement.
> 
> Debian (testing) is up to v4.4.0 so that would also be suitable: https://packages.debian.org/search?keywords=sdcc

Yes, it is an issue to be observed, the time that the version already has. I will try to install version 4.5.0 for Linux:

https://sourceforge.net/projects/sdcc/files/sdcc-linux-amd64/4.5.0/

rtek1000, 2025-02-07:
> I will try to install version 4.5.0 for Linux:
> 
> https://sourceforge.net/projects/sdcc/files/sdcc-linux-amd64/4.5.0/

It was easy, it has simple instructions:
```
SDCC Binary Kit Install Help
----------------------------

For Linux users:
================
To install:

* Extract the binary kit to a temporary directory.
  This will create a new directory called 'sdcc-4.5.0' in the temporary directory.
    cd ~
    mkdir tmp
    cd tmp
    tar xjf path/to/binary/kit/sdcc-4.5.0-amd64-unknown-linux2.5.tar.bz2

* Change to the sdcc directory and copy all files to /usr/local
    cd sdcc-4.5.0
    cp -r * /usr/local

This will install sdcc binaries into: /usr/local/bin/
header files into:                    /usr/local/share/sdcc/include/
non-free header files into:           /usr/local/share/sdcc/non-free/include/
library files into:                   /usr/local/share/sdcc/lib/
non-free library files into:          /usr/local/share/sdcc/non-free/lib/
and documentation into:               /usr/local/share/sdcc/doc/

You can test the install by entering:
    /usr/local/bin/sdcc -v

This should return sdcc's version number.

Even if the default search paths look into /usr/local, sdcc is fully
relocatable. Try `sdcc --print-search-dirs` if you have problems with header
or library files. See sdccman.pdf for more detailed information.
```

rtek1000, 2025-02-07:
I found in the hal.c file:

[`pg. 43, sec. 8.4.1.2.`](https://github.com/mightymos/RF-Bridge-OB38S003/blob/a09420a5652cd8b53280e907c4c56415dcf406f1/drivers/ob38s003/src/hal.c#L112)

Where can I find these documents?

mightymos, 2025-02-07:
> I found in the hal.c file:
> 
> [`pg. 43, sec. 8.4.1.2.`](https://github.com/mightymos/RF-Bridge-OB38S003/blob/a09420a5652cd8b53280e907c4c56415dcf406f1/drivers/ob38s003/src/hal.c#L112)
> 
> Where can I find these documents?
> 
> P.S.: OnbrightFlasher did not recognize the microcontroller:
> 
> ```
> 2025-02-07 11:39:44,452 - INFO - cycle power to target (start with power off and then turn on)
> 2025-02-07 11:39:54,816 - INFO - Handshake succeeded
> 2025-02-07 11:39:54,916 - INFO - Status: 3
> 2025-02-07 11:39:54,919 - INFO - NACK on transmit of data
> 2025-02-07 11:39:54,920 - INFO - Chip read type FAILED
> 2025-02-07 11:39:54,922 - INFO - Chip type reported was: 0xD0
> 2025-02-07 11:39:54,949 - INFO - Can try command [signature] or [idle] then [handshake] to retry
> 2025-02-07 11:39:54,950 - ERROR - Something went wrong. Process ended with Can try command [signature] or [idle] then [handshake] to retry
> 2025-02-07 11:39:54,950 - ERROR - State 4 failed. Returning to handshake state.
> ```

I linked the datasheet in this comment:
https://github.com/mightymos/RF-Bridge-OB38S003/issues/29#issuecomment-2638755070

It seems like when people have problems with the flasher it is usually an issue with their power supply.
The power supply for the target needs to supply enough current.
Instead of using the script you can try using the Arduino IDE Serial Monitor to manually issue erase commands to wipe the stock firmware to see if anything succeeds.

You can then try reading the chip type.

It is difficult for me to troubleshoot remotely.

rtek1000, 2025-02-08:
Apparently the NodeMCU managed to erase the chip.

```
handshake
State changing to handshake
cycle power to target (start with power off and then turn on)

Handshake succeeded
Status: 3
NACK on transmit of data
Chip read type FAILED
Chip type reported was: 0xD0
Can try command [signature] or [idle] then [handshake] to retry
erase
Erasing chip...
Status: 0
Success
Chip erase successful
```

When I read the instruction to leave the Chip connected to the Programmer, and then turn on the power when prompted, I did not consider the power that the I2C pull-up resistors deliver to the Chip.

This time when Status 3 appeared, I left everything disconnected and turned everything on at once.

It may work better if you leave the I2C pull-up resistors connected to the Chip's power supply, so that synchronization with the Chip's power supply occurs.

rtek1000, 2025-02-08:
It worked, I could see pin 8 of the OB38S003 Chip toggling on the oscilloscope. 

```
python flashScript.py 
Available COM ports:
1. /dev/ttyS0 - n/a
...
32. /dev/ttyS31 - n/a
33. /dev/ttyUSB0 - USB2.0-Serial
Select a COM port (enter the number): 33
2025-02-07 22:51:05,453 - INFO - Serial connection opened on /dev/ttyUSB0
Select a file by number:
1. blink.ihx
Enter the number of the file you want to select: 1
2025-02-07 22:51:19,681 - INFO - You selected: /media/.../OB38S003/OnbrightFlasher-main/blink.ihx
2025-02-07 22:51:20,785 - INFO - handshake
2025-02-07 22:51:20,787 - INFO - State changing to handshake
2025-02-07 22:51:20,789 - INFO - cycle power to target (start with power off and then turn on)
2025-02-07 22:51:42,759 - INFO - Handshake succeeded
2025-02-07 22:51:42,878 - INFO - Status: 0
2025-02-07 22:51:42,880 - INFO - Success
2025-02-07 22:51:42,881 - INFO - Chip read: 0xA
2025-02-07 22:51:42,882 - INFO - Connected...
2025-02-07 22:51:42,884 - INFO - Returning to idle state...
2025-02-07 22:51:42,986 - INFO - erase
2025-02-07 22:51:43,488 - INFO - Erasing chip...
2025-02-07 22:51:43,489 - INFO - Status: 0
2025-02-07 22:51:43,490 - INFO - Success
2025-02-07 22:51:43,492 - INFO - Chip erase successful
2025-02-07 22:51:43,593 - INFO - 
2025-02-07 22:51:44,094 - INFO - setfuse 18 249
2025-02-07 22:51:44,596 - INFO - Set configuration byte...
2025-02-07 22:51:44,597 - INFO - Status: 0
2025-02-07 22:51:44,598 - INFO - Success
2025-02-07 22:51:44,600 - INFO - Wrote configuration byte
2025-02-07 22:51:44,606 - INFO - :03000000020006F5
2025-02-07 22:51:44,609 - INFO - Write successful
2025-02-07 22:51:44,611 - INFO - Wrote 3 bytes
2025-02-07 22:51:44,616 - INFO - :03005F0002000399
2025-02-07 22:51:44,620 - INFO - Write successful
2025-02-07 22:51:44,621 - INFO - Wrote 3 bytes
2025-02-07 22:51:44,626 - INFO - :0300030002008672
2025-02-07 22:51:44,630 - INFO - Write successful
2025-02-07 22:51:44,632 - INFO - Wrote 3 bytes
2025-02-07 22:51:44,642 - INFO - :20006200AE82AF837C007D00C3EC9EED9F50147AE87B031ABAFF011BEA4B70F70CBC00E8CB
2025-02-07 22:51:44,674 - INFO - Write successful
2025-02-07 22:51:44,675 - INFO - Wrote 32 bytes
2025-02-07 22:51:44,684 - INFO - :1C0082000D80E52253DBFE43DA01C2B09001F4120062D2B09001F412006280EE30
2025-02-07 22:51:44,713 - INFO - Write successful
2025-02-07 22:51:44,714 - INFO - Wrote 28 bytes
2025-02-07 22:51:44,721 - INFO - :06003500E478FFF6D8FD9F
2025-02-07 22:51:44,727 - INFO - Write successful
2025-02-07 22:51:44,729 - INFO - Wrote 6 bytes
2025-02-07 22:51:44,739 - INFO - :200013007900E94400601B7A009000A2780175A000E493F2A308B8000205A0D9F4DAF275F1
2025-02-07 22:51:44,772 - INFO - Write successful
2025-02-07 22:51:44,773 - INFO - Wrote 32 bytes
2025-02-07 22:51:44,779 - INFO - :02003300A0FF2C
2025-02-07 22:51:44,782 - INFO - Write successful
2025-02-07 22:51:44,783 - INFO - Wrote 2 bytes
2025-02-07 22:51:44,793 - INFO - :20003B007800E84400600A790175A000E4F309D8FC7800E84400600C7900900001E4F0A3C3
2025-02-07 22:51:44,826 - INFO - Write successful
2025-02-07 22:51:44,827 - INFO - Wrote 32 bytes
2025-02-07 22:51:44,833 - INFO - :04005B00D8FCD9FAFA
2025-02-07 22:51:44,837 - INFO - Write successful
2025-02-07 22:51:44,839 - INFO - Wrote 4 bytes
2025-02-07 22:51:44,846 - INFO - :0D00060075810712009EE582600302000371
2025-02-07 22:51:44,859 - INFO - Write successful
2025-02-07 22:51:44,861 - INFO - Wrote 13 bytes
2025-02-07 22:51:44,867 - INFO - :04009E007582002245
2025-02-07 22:51:44,871 - INFO - Write successful
2025-02-07 22:51:44,873 - INFO - Wrote 4 bytes
2025-02-07 22:51:44,877 - INFO - :00000001FF
2025-02-07 22:51:44,979 - INFO - mcureset
2025-02-07 22:51:45,481 - INFO - MCU reset...
Serial connection closed.
```

Now I need to isolate the Timer and UART codes, in simple programs, to understand how to use them.

rtek1000, 2025-02-08:
It may be interesting for users to know about the small write capacity of the ROM (program) memory, this OB38S003 has only 1'000 guaranteed writes. An Arduino UNO (Atmega328) has 10'000. ESp8266 has 100'000.

Depending on the complexity of the program, it is better to do the tests on Arduino (or NodeMCU, ESP32, etc.), only after it is bug-free, burn the OB38S003.

[image]

rtek1000, 2025-02-10:
Hello, I managed to get almost everything to work, I couldn't get the RESET P1.5 pin to work, do I have to change a fuse?

-----

Regarding UART, to be able to send data, I needed to unlink TI and RI in the data reception interrupt handling routine, because they have the same interrupt vector.

-----

I also found the official programmer msm9066 at an attractive price, in case anyone wants to try it:

https://pt.aliexpress.com/item/1005007667412447.html

rtek1000, 2025-02-10:
If anyone wants to continue with this project, I left the modified files and other tips I found (such as example codes provided by the manufacturer) in this repository linked below, for now I will leave this issue closed, thank you.

https://github.com/rtek1000/OB38S003-PANTILT-UNIT-FOR-CAMERA

[image]

closed 2025-02-10 by rtek1000

mightymos, 2025-02-10:
> > Hello, I managed to get almost everything to work, I couldn't get the RESET P1.5 pin to work, do I have to change a fuse?
> 
> Ok, I managed to make the P1.5 pin work as GPIO, but unfortunately now it's too late, the MCU had a burnt pin.
> 
> I found that by default OnbrightFlasher seems to be writing fuse 18 249, and so pin P1.5 operates as RESET input.
> 
> To make the RESET pin work as GPIO:
> 
>     * I used the Arduino serial monitor, and ran the command "setfuse 18 1" (after programming the MCU with my code)
> 
> 
> > version                                                            <----- typed manually
> > Date: Feb  7 2025 Time: 10:41:26
> > handshake                                                            <----- typed manually
> > State changing to handshake
> > cycle power to target (start with power off and then turn on)
> > Handshake succeeded
> > Status: 0
> > Success
> > Chip read: 0xA
> > Connected...
> > Returning to idle state...
> > fuse 18                                                            <----- typed manually
> > Get configuration byte...
> > Status: 0
> > Success
> > Configuration byte at (18) is: 249
> > setfuse 18 1                                                            <----- typed manually
> > Set configuration byte...
> > Status: 0
> > Success
> > Wrote configuration byte
> > mcureset                                                            <----- typed manually

Oh no, sorry about the damaged pin.
But glad you made some progress and thank you for finding the less expensive MSM9066.