mightymos/RF-Bridge-OB38S003 #31: Bucket sniffing for Acurite 0592TXR sensor

andrew-p0, 2025-02-16:
It seems that I cant get bucket sniffing working for Acurite 592TXR Temperature / Humidity sensor. I can capture the signal with logic analyzer connected to pin 5 of U4 on Sonoff R2 v2.2 bridge. The measurements of the capture are follow.

SYNC = 4 repeats HIGH(650us) LOW(550us)
BIT_1 = HIGH(440us) LOW (180us)
BIT_0 = HIGH(240us) LOW(360us)
END = HIGH(240) LOW(2160us)

The whole sequence is repeated 3 times. I have added following buckets: { 600, 200, 420, 2000 }, but got nothing.
Also I have tried to play with TOLERANCE_MIN/MAX, but it didn't make much difference.

mightymos, 2025-02-16:
I can double check but I do not think the sniffing logic can handle multiple sync bits.
If I am incorrect I will try to check and let you know.

But it might be easier to use an Acurite specific decoding project:
https://github.com/merbanan/rtl_433/blob/master/src/devices/acurite.c
https://github.com/swoboda1337/acurite-esphome

mightymos, 2025-02-24:
Do you know how many data bits you have?
I'm wondering if we could add your protocol to portisch_protocols.h and decode with 0xA6 command.

mightymos, 2025-02-24:
Interesting, someone tried to decode on original Portisch also:
https://github.com/Portisch/RF-Bridge-EFM8BB1/issues/121

Are you sure about the format?
That user seems to discuss a different format there but I am not sure.

andrew-p0, 2025-02-25:
> Interesting, someone tried to decode on original Portisch also: [Portisch/RF-Bridge-EFM8BB1#121](https://github.com/Portisch/RF-Bridge-EFM8BB1/issues/121)
> 
> Are you sure about the format? That user seems to discuss a different format there but I am not sure.

Here is the capture that I did: [rtl_433 capture](https://triq.org/pdv/#AAB04905020268019000D808982714808080809292A1929292A1A1A1A1A1A1A192A192A192A1A1A192A1A19292A1A1A192A19292A1A1A192A1A1A19292929292A192A1A19292A19292A1A1A355+AAB04905010268019000D808982714808080809292A1929292A1A1A1A1A1A1A192A192A192A1A1A192A1A19292A1A1A192A19292A1A1A192A1A1A19292929292A192A1A19292A19292A1A1A455). The timings that you have referenced is pretty accurate

mightymos, 2025-02-25:
That's a neat signal analysis tool.
So I think this is a breakdown of the 0xB0 command:

```
AAB0C3
// # of buckets
05
// repeats
01
// bucket timings
0268
0190
00D8
0898
2714

// sync then data
80808080
9292A1929292A1A1A1A1A1A1A192A192A192A1A1A192A1A19292A1A1A192A19292A1A1A192A1A1A19292929292A192A1A19292A19292A1A1A38

0808080
9292A1929292A1A1A1A1A1A1A192A192A192A1A1A192A1A19292A1A1A192A19292A1A1A192A1A1A19292929292A192A1A19292A19292A1A1A38

080808
09292A1929292A1A1A1A1A1A1A192A192A192A1A1A192A1A19292A1A1A192A19292A1A1A192A1A1A19292929292A192A1A19292A19292A1A1A4

// end of packet
55
```

So in other words, the data is being repeated and so it's making these really long 0xB1 or 0xB0 commands instead of just specifying more repeats.

I guess my question is what are you ultimately trying to do?
Just decode the data?

mightymos, 2025-02-25:
Ok, can you post your modified portisch_protocols.h here?
Maybe I can check it to see if there is a mistake for the 0592TXR.

andrew-p0, 2025-02-26:
Here is my changes to portisch_protocols.h. ACU606TX works, but ACU592TXR does not. I was reading the data with the  advanced sniffing activated.
The beginning of the file show differences probably because of the line endings.
```c
diff --git a/inc/portisch_protocols.h b/inc/portisch_protocols.h
index 88dacd1..c552e2b 100644
--- a/inc/portisch_protocols.h
+++ b/inc/portisch_protocols.h
@@ -10,10 +10,10 @@
 
 //-----------------------------------------------------------------------------
 // Includes
-//-----------------------------------------------------------------------------
+//-----------------------------------------------------------------------------
 #include <stddef.h>
-#include <stdint.h>
-
+#include <stdint.h>
+
 #include "portisch_macros.h"
 #include "portisch_config.h"
 
@@ -38,10 +38,10 @@
 
 typedef struct PROTOCOL_STATUS
 {
-	//uint16_t status;
-	uint8_t sync_status;
-	uint8_t bit0_status;
-	uint8_t bit1_status;
+	//uint16_t status;
+	uint8_t sync_status;
+	uint8_t bit0_status;
+	uint8_t bit1_status;
 	uint8_t end_status;
 	uint8_t bit_count;
 	uint8_t actual_bit_of_byte;
@@ -324,6 +324,31 @@ __code static uint8_t PROTOCOL_BIT0(H13726)[]  = { LOW(1), HIGH(0) };
 __code static uint8_t PROTOCOL_BIT1(H13726)[]  = { LOW(2), HIGH(0) };
 #endif
 
+/*
+ * Acurite 606TX Temperature sensor
+ */
+#if defined(PORTISCH_SUPPORT_ACU606TX_PROTOCOL)
+#define ACU606TX
+__code static uint16_t PROTOCOL_BUCKETS(ACU606TX)[] = { 500, 1950, 3900, /*9150*/ 8850 };
+__code static uint8_t PROTOCOL_START(ACU606TX)[] = { /*HIGH(0), LOW(0), HIGH(0), LOW(0),*/ HIGH(0), LOW(3) };
+__code static uint8_t PROTOCOL_BIT0(ACU606TX)[]  = { HIGH(0), LOW(1) };
+__code static uint8_t PROTOCOL_BIT1(ACU606TX)[]  = { HIGH(0), LOW(2) };
+/*__code static uint8_t PROTOCOL_END(ACU606TX)[]   = { HIGH(0), LOW(3) };*/
+#endif
+
+/*
+ * Acurite 592TXR Temperature Humidity sensor
+ */
+#if defined(PORTISCH_SUPPORT_ACU592TXR_PROTOCOL)
+#define ACU592TXR
+__code static uint16_t PROTOCOL_BUCKETS(ACU592TXR)[] = { 600, 200, 400, 2200 };
+__code static uint8_t PROTOCOL_START(ACU592TXR)[] = { HIGH(0), LOW(0), HIGH(0), LOW(0),
+						       HIGH(0), LOW(0), HIGH(0), LOW(0) };
+__code static uint8_t PROTOCOL_BIT0(ACU592TXR)[]  = { HIGH(1), LOW(2) };
+__code static uint8_t PROTOCOL_BIT1(ACU592TXR)[]  = { HIGH(2), LOW(1) };
+__code static uint8_t PROTOCOL_END(ACU592TXR)[]   = { HIGH(1), LOW(3) };
+#endif
+
 __code static struct BUCKET_PROTOCOL_DATA PROTOCOL_DATA[] =
 {
 #if defined(PORTISCH_SUPPORT_PT226X_PROTOCOL)
@@ -599,8 +624,35 @@ __code static struct BUCKET_PROTOCOL_DATA PROTOCOL_DATA[] =
 			36
 		},
 #endif
+#if defined(PORTISCH_SUPPORT_ACU606TX_PROTOCOL)
+		/*
+		 * Acurite 606TX Temperature
+		 */
+		{
+			{ &PROTOCOL_BUCKETS(ACU606TX)[0], ARRAY_LENGTH(PROTOCOL_BUCKETS(ACU606TX)) },
+			{ &PROTOCOL_START(ACU606TX)[0], ARRAY_LENGTH(PROTOCOL_START(ACU606TX)) },
+			{ &PROTOCOL_BIT0(ACU606TX)[0], ARRAY_LENGTH(PROTOCOL_BIT0(ACU606TX)) },
+			{ &PROTOCOL_BIT1(ACU606TX)[0], ARRAY_LENGTH(PROTOCOL_BIT1(ACU606TX)) },
+			/*{ &PROTOCOL_END(ACU606TX)[0], ARRAY_LENGTH(PROTOCOL_END(ACU606TX)) },*/
+			{ NULL, 0 },
+			32
+		},
+#endif
+#if defined(PORTISCH_SUPPORT_ACU592TXR_PROTOCOL)
+		/*
+		 * Acurite 592TXR Temperature Humidity sensor
+		 */
+		{
+			{ &PROTOCOL_BUCKETS(ACU592TXR)[0], ARRAY_LENGTH(PROTOCOL_BUCKETS(ACU592TXR)) },
+			{ &PROTOCOL_START(ACU592TXR)[0], ARRAY_LENGTH(PROTOCOL_START(ACU592TXR)) },
+			{ &PROTOCOL_BIT0(ACU592TXR)[0], ARRAY_LENGTH(PROTOCOL_BIT0(ACU592TXR)) },
+			{ &PROTOCOL_BIT1(ACU592TXR)[0], ARRAY_LENGTH(PROTOCOL_BIT1(ACU592TXR)) },
+			{ &PROTOCOL_END(ACU592TXR)[0], ARRAY_LENGTH(PROTOCOL_END(ACU592TXR)) },
+			56
+		},
+#endif
 };
-
+
 // https://www.ashn.dev/blog/2020-01-06-c-array-length.html
 #define NUM_OF_PROTOCOLS (sizeof(PROTOCOL_DATA) / sizeof(PROTOCOL_DATA[0]))
 #endif // INC_RF_PROTOCOLS_H_
```

mightymos, 2025-02-26:
Based on comparing to your plot, for 592 should the bits be instead?
bit 0: HIGH(2) LOW(1)
bit 1: HIGH(1) LOW(2)

andrew-p0, 2025-02-26:
It seems that decoded data in the plot is inverted. I did manual decoding according to rtl_433 source code and it matches when bits are inverted.

mightymos, 2025-02-26:
Ok, I'm glad it's working, even if not reliably.
I am not sure right now what could be the problem.

I may try to simulate the transmitted signal to see if I can decode it also.
But could you also post a link to the product/weather station model you bought?

andrew-p0, 2025-02-27:
> I may try to simulate the transmitted signal to see if I can decode it also. But could you also post a link to the product/weather station model you bought?

I have following two that I've got from HomeDepot a while ago:
[Acurite 02014HDSB](https://support.acurite.com/hc/en-us/articles/4463667811995-02010-Weather-Station)
[Acurite 00827](https://support.acurite.com/hc/en-us/articles/27646453937819-00827-How-to-Dress-Weather-Forecaster)

mightymos, 2025-03-10:
> > I may try to simulate the transmitted signal to see if I can decode it also. But could you also post a link to the product/weather station model you bought?
> 
> I have following two that I've got at HomeDepot a while ago: [Acurite 02014HDSB](https://support.acurite.com/hc/en-us/articles/4463667811995-02010-Weather-Station) [Acurite 00827](https://support.acurite.com/hc/en-us/articles/27646453937819-00827-How-to-Dress-Weather-Forecaster)

Okay it does not seem like these are still available for purchase.
I am not sure when I will ever be able to look into this specific protocol.

andrew-mtk, 2025-03-21:
You cold try to check [Model # 00384HD](https://www.homedepot.com/p/AcuRite-Digital-Color-Display-Wireless-Indoor-Outdoor-Thermometer-00384HD/311697358#overlay). It is on clearance at HD.

mightymos, 2025-03-22:
> You cold try to check [Model # 00384HD](https://www.homedepot.com/p/AcuRite-Digital-Color-Display-Wireless-Indoor-Outdoor-Thermometer-00384HD/311697358#overlay). Some locations has is on clearance.

Ok, I may not be able to help much further.
It's just a lack of time.

Did you get any working output ever with Tasmota for example?
If so could you post the output?

This person got another weather station partially working so might be able to consult with them:
https://github.com/mightymos/RF-Bridge-OB38S003/issues/34

phoenix-9, 2025-10-27:
[I use a different protocol that looks like this.](https://triq.org/pdv/#AAB0160401031C01901430271481919191919191919191919255+AAB04C0401031C01901430271481819090908190819090819081819090908181909090908181818181909081908181909081909090819090819090909090909090818181908181818181818181909355)
Twelve sync pulses, a gap, and 66 bits of data.  I got it to decode and transmit successfully with this change to the source:

```
/*
 * Hunter/Casablanca ceiling fans
 * FCC ID IN2TX47 (transmitter) and IN2TX46 (receiver)
 * This is similar to HCS200/HCS300 protocol, see rtl_433 project
 * but without rolling codes.  12 sync pulses.  66 bits data.
 * Device ID is transmitted first.  Then the data.
 */
#if defined(PORTISCH_SUPPORT_HUNTER_PROTOCOL)
#define HUNTER
__code static uint16_t PROTOCOL_BUCKETS(HUNTER)[] = { 400, 800, 5100, 10400 };
__code static uint8_t PROTOCOL_START(HUNTER)[] = { HIGH(0), LOW(0), HIGH(0), LOW(0), HIGH(0), LOW(0),
                                                   HIGH(0), LOW(0), HIGH(0), LOW(0), HIGH(0), LOW(0),
                                                   HIGH(0), LOW(0), HIGH(0), LOW(0), HIGH(0), LOW(0),
                                                   HIGH(0), LOW(0), HIGH(0), LOW(0), HIGH(0), LOW(2) };
__code static uint8_t PROTOCOL_BIT0(HUNTER)[]  = { HIGH(1), LOW(0) };
__code static uint8_t PROTOCOL_BIT1(HUNTER)[]  = { HIGH(0), LOW(1) };
__code static uint8_t PROTOCOL_END(HUNTER)[]   = { LOW(3) };
#endif
```
and
```
#if defined(PORTISCH_SUPPORT_HUNTER_PROTOCOL)
		/*
		 * Hunter/Casablanca ceiling fans
		 */
		{
			{ &PROTOCOL_BUCKETS(HUNTER)[0], ARRAY_LENGTH(PROTOCOL_BUCKETS(HUNTER)) },
			{ &PROTOCOL_START(HUNTER)[0], ARRAY_LENGTH(PROTOCOL_START(HUNTER)) },
			{ &PROTOCOL_BIT0(HUNTER)[0], ARRAY_LENGTH(PROTOCOL_BIT0(HUNTER)) },
			{ &PROTOCOL_BIT1(HUNTER)[0], ARRAY_LENGTH(PROTOCOL_BIT1(HUNTER)) },
			{ &PROTOCOL_END(HUNTER)[0], ARRAY_LENGTH(PROTOCOL_END(HUNTER)) },
			66
		},
#endif
```