mightymos/RF-Bridge-OB38S003 #32: ESPHome Transmit Issues

joelguth, 2025-02-17:
Hello,

First off, thanks for all of the effort on this project! 

I bought an rf bridge recently to control some cheap LED strips. I've spent a few days now flashing various firmware, and I've run into some reproducible issues.

The first issue i'm facing is transmitting in ESPhome. Even with passthrough firmware i don't seem to be able to reliably transmit anything, particularly with the rc switch implementation in ESPhome (Which in my opinion seems very broken). 

I have re-flashed the rf bridge with Tasmota and the passthrough firmware, and it works flawlessly. Captured codes are able to be re-transmitted without any issues. 

For example, i was able to capture the below using Tasmota with the passthrough firmware:
`RfReceived":{"Data":"0xAAF01","Bits":24,"Protocol":1,"Pulse":393}`

And then I re-transmitted that same data using protocol 1 with that pulse length with Tasmota and succeeded. 

I prefer ESPHome over Tasmota for the integration with Home Assistant, so i was hoping just flash ESPHome back and re-transmit the codes i captured (Again, in ESPhome receiving just incorrectly gives protocol 1 without a pulse length).

I tried to then re-transmit what i captured above using ESPHome, and it is once again not succeeding:
`switch:
  - platform: template
    name: "Mirror Power"
    turn_on_action:
      - remote_transmitter.transmit_rc_switch_raw:
          code: '10101010111100000001'
          protocol:
            pulse_length: 393
            inverted: False`

Here is my ESPHome Transmitter config:
`remote_transmitter:
  pin: GPIO1
  carrier_duty_percent: 100%`

I have a flipper zero nearby that is able to capture these remote codes as well, and it does not seem to detect that "0xAAF01" code being sent from ESPHome.

Side Note: For switching between ESPHome to Tasmota, you have to erase the radio firmware (at least in the case of the passthrough firmware) to flash the ESP. There must be too much interference on the serial lines to reliably flash without?

I also tried your portisch port with ESPhome, and i seemed to be able to reliably capture codes but when they were re-transmitted they again did not seem successful. Maybe there is just something goofy with the ESPHome rf bridge and rf transmitter implementations.

Does anyone else seem to be experiencing similar issues? Have others had greater success with Tasmota on these rf bridges than ESPHome?

Thank you!

mightymos, 2025-02-17:
It looks like your code 0xAAF01 is only 20 bits (4 bits * 5 characters) but standard protocols is 24 bits.
Maybe the print out just did not display the leading zero.

So I think your code might be 0x0AAF01.
I think you can try transmitting that to see if it works in place of your remote.

When you press different buttons on your remote are the codes similar except the last byte changes (i.e., a single bit is changing)?

I think the base pulse timing is probably 350 microseconds. 
It's possible it is actually the measured value 390 microseconds but I cannot be sure.

You don't necessarily need to specify the time.
You can just specify the protocol ID and rcswitch will look up timings from a table.

mightymos, 2025-02-17:
You can just do this it looks like:

```
on_...:
  - remote_transmitter.transmit_rc_switch_raw:
      code: 'file me in'
      protocol: 1
```

joelguth, 2025-02-17:
Interesting, the flipper does pick up the code as 0x0AAF01 and detected it as "Princeton 24 Bit", so good catch. Interesting that that still worked with Tasmota when I retransmitted in that same format.

It's definitely closer to the 390 range, the flipper is also picking up a time domain around 390-400, which I don't think aligns with protocol 1. I assume in this case I need to specify the pulse length.

The codes do seem similar, i.e. the Red button is 0xAAF10 (0x0AAF10), blue is 0xAAF11 (0xAAF11).

mightymos, 2025-02-19:
> And testing with:
> 
> ```
>     turn_on_action:
>       - remote_transmitter.transmit_rc_switch_raw:
>           code: '00010101010111100000001'
>           protocol: 1
> ```
> 

That only shows 23 bits, but you need 24 bits.
You can count with this tool:
https://mothereff.in/byte-counter

Shouldn't there be four zeros at the front?

joelguth, 2025-02-19:
Understood, I'll give portisch another go. I think this particular protocol was one of the advanced protocols and I had some issues sniffing those with portisch, but now that I know the codes I can try an advanced transmit with the newer release you have supporting the advanced protocols.

Worst case i'll fall back to Tasmota with passthrough, as that worked for me.

I appreciate your help on this!

joelguth, 2025-02-19:
That would require the bucket dump capabilities, correct? So I should install the "bucket sniffing included" release?

mightymos, 2025-02-19:
Yes. Hopefully it will do what I think we need.

joelguth, 2025-02-19:
Alright well, Portisch seems much more successful in terms of transmit. The code is at least properly transmitting now. For fun I tested with both ESPhome and Tasmota and was successful with both, though I'm very underwhelmed by the transmit distance on this board. I'm only able to pick up the code from around 25 feet away.  Maybe my board is just a dud, but I expected a bit more range.

Anyways, i think we can consider this resolved. I think the tldr is that the ESPhome rc switch implementation with the passthrough firmware is not great. It did however work pretty well with Tasmota. Portisch seems to work fine on both.

Thank you!

mightymos, 2025-02-19:
Admittedly I'm not too familiar with what limits the transmission range.
But glad it is able to replay codes captured from the remote.

Closing for now since issue was resolved.

closed 2025-02-19 by mightymos