mightymos/RF-Bridge-OB38S003 #33: Syntax error in compilation of tag v0.4.13

G-Tarik, 2025-02-21:
Hello @mightymos , great work you are doing here and thank you very much for the effort of porting Portisch firmware to sdcc.  
I have a small issue and I believe you will quickly identify the cause.

I cloned the repo and wanted to compile to be sure my dev setup works  and that I can reproduce the same files as you released in v0.4.13. But it failed with error.
```bash
$ git status
HEAD detached at v0.4.13
nothing to commit, working tree clean

$ git log -1
commit c59e4da17d41feb01e115bbe64ceb083e5a2f34a (HEAD, tag: v0.4.13)
Author: Jonathan Armstrong <jda2158@gmail.com>
Date:   Tue Jan 28 01:20:33 2025 -0600

    support conditional compiliation

$ make
Compiling src/main_passthrough.c
mkdir -p object/
sdcc -mmcs51 --model-small  -DTARGET_BOARD_OB38S003 -DBUCKET_SNIFFING_INCLUDED -DMCU_FREQ=16000000UL -Iinc/ -Idrivers/ob38s003/inc -c -o object/main_passthrough.rel src/main_passthrough.c
Compiling drivers/ob38s003/src/delay.c
mkdir -p object/
sdcc -mmcs51 --model-small  -DTARGET_BOARD_OB38S003 -DBUCKET_SNIFFING_INCLUDED -DMCU_FREQ=16000000UL -Iinc/ -Idrivers/ob38s003/inc -c -o object/delay.rel drivers/ob38s003/src/delay.c
Compiling drivers/ob38s003/src/hal.c
mkdir -p object/
sdcc -mmcs51 --model-small  -DTARGET_BOARD_OB38S003 -DBUCKET_SNIFFING_INCLUDED -DMCU_FREQ=16000000UL -Iinc/ -Idrivers/ob38s003/inc -c -o object/hal.rel drivers/ob38s003/src/hal.c
Linking object/main_passthrough.rel object/delay.rel object/hal.rel
mkdir -p build/
sdcc -mmcs51 --model-small --iram-size 256 --xram-size 256 --code-size 8192 -o build/main_passthrough_OB38S003.ihx object/main_passthrough.rel object/delay.rel object/hal.rel
# hex lines are a short, fixed length (compared with ihx) and therefore works with upload tools
packihx build/main_passthrough_OB38S003.ihx > build/main_passthrough_OB38S003.hex
packihx: read 22 lines, wrote 34: OK.
# unix style line endings (LF instead of LFCR) work with upload tools
dos2unix build/main_passthrough_OB38S003.hex
dos2unix: converting file build/main_passthrough_OB38S003.hex to Unix format...
Compiling src/main_portisch.c
mkdir -p object/
sdcc -mmcs51 --model-small  -DTARGET_BOARD_OB38S003 -DBUCKET_SNIFFING_INCLUDED -DMCU_FREQ=16000000UL -Iinc/ -Idrivers/ob38s003/inc -c -o object/main_portisch.rel src/main_portisch.c
src/main_portisch.c:417: syntax error: token -> 'uint8_t' ; column 27
src/main_portisch.c:428: error 20: Undefined identifier 'num_buckets'
src/main_portisch.c:432: error 20: Undefined identifier 'num_buckets'
src/main_portisch.c:432: error 20: Undefined identifier 'num_buckets'
make: *** [Makefile:235: object/main_portisch.rel] Error 1
``` 

Expected behavior: release tag compiles without issues.

Just in case:
```bash
$ uname -a
Linux e2d26e6e9daa 6.5.0-35-generic #35~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Tue May  7 09:00:52 UTC 2 x86_64 x86_64 x86_64 GNU/Linux

$ sdcc --version
SDCC : mcs51/z80/z180/r2k/r3ka/gbz80/tlcs90/ez80_z80/ds390/TININative/ds400/hc08/s08/stm8/pdk13/pdk14/pdk15 4.0.0 #11528 (Linux)
published under GNU General Public License (GPL)
```

Then I tried `main` branch, still error but different:
```bash
$ make
Compiling src/passthrough_main.c
mkdir -p object/
sdcc -mmcs51 --model-small  -DTARGET_MCU_OB38S003 -DBUCKET_SNIFFING_INCLUDED -DMCU_FREQ=16000000UL -I. -Iinc/ -Idrivers/ob38s003/inc -c -o object/passthrough_main.rel src/passthrough_main.c
Compiling drivers/ob38s003/src/delay.c
mkdir -p object/
sdcc -mmcs51 --model-small  -DTARGET_MCU_OB38S003 -DBUCKET_SNIFFING_INCLUDED -DMCU_FREQ=16000000UL -I. -Iinc/ -Idrivers/ob38s003/inc -c -o object/delay.rel drivers/ob38s003/src/delay.c
Compiling drivers/ob38s003/src/hal.c
mkdir -p object/
sdcc -mmcs51 --model-small  -DTARGET_MCU_OB38S003 -DBUCKET_SNIFFING_INCLUDED -DMCU_FREQ=16000000UL -I. -Iinc/ -Idrivers/ob38s003/inc -c -o object/hal.rel drivers/ob38s003/src/hal.c
Linking object/passthrough_main.rel object/delay.rel object/hal.rel
mkdir -p build/
sdcc -mmcs51 --model-small --iram-size 256 --xram-size 256 --code-size 8192 -o build/passthrough_main_OB38S003.ihx object/passthrough_main.rel object/delay.rel object/hal.rel
# hex lines are a short, fixed length (compared with ihx) and therefore works with upload tools
packihx build/passthrough_main_OB38S003.ihx > build/passthrough_main_OB38S003.hex
packihx: read 22 lines, wrote 34: OK.
# unix style line endings (LF instead of LFCR) work with upload tools
dos2unix build/passthrough_main_OB38S003.hex
dos2unix: converting file build/passthrough_main_OB38S003.hex to Unix format...
Compiling src/portisch_main.c
mkdir -p object/
sdcc -mmcs51 --model-small  -DTARGET_MCU_OB38S003 -DBUCKET_SNIFFING_INCLUDED -DMCU_FREQ=16000000UL -I. -Iinc/ -Idrivers/ob38s003/inc -c -o object/portisch_main.rel src/portisch_main.c
Compiling src/portisch_rf_handling.c
mkdir -p object/
sdcc -mmcs51 --model-small  -DTARGET_MCU_OB38S003 -DBUCKET_SNIFFING_INCLUDED -DMCU_FREQ=16000000UL -I. -Iinc/ -Idrivers/ob38s003/inc -c -o object/portisch_rf_handling.rel src/portisch_rf_handling.c
Compiling src/portisch_serial.c
mkdir -p object/
sdcc -mmcs51 --model-small  -DTARGET_MCU_OB38S003 -DBUCKET_SNIFFING_INCLUDED -DMCU_FREQ=16000000UL -I. -Iinc/ -Idrivers/ob38s003/inc -c -o object/portisch_serial.rel src/portisch_serial.c
Compiling drivers/ob38s003/src/timer_interrupts.c
mkdir -p object/
sdcc -mmcs51 --model-small  -DTARGET_MCU_OB38S003 -DBUCKET_SNIFFING_INCLUDED -DMCU_FREQ=16000000UL -I. -Iinc/ -Idrivers/ob38s003/inc -c -o object/timer_interrupts.rel drivers/ob38s003/src/timer_interrupts.c
Compiling src/uart.c
mkdir -p object/
sdcc -mmcs51 --model-small  -DTARGET_MCU_OB38S003 -DBUCKET_SNIFFING_INCLUDED -DMCU_FREQ=16000000UL -I. -Iinc/ -Idrivers/ob38s003/inc -c -o object/uart.rel src/uart.c
Linking object/portisch_main.rel object/portisch_rf_handling.rel object/portisch_serial.rel object/timer_interrupts.rel object/uart.rel object/hal.rel
mkdir -p build/
sdcc -mmcs51 --model-small --iram-size 256 --xram-size 256 --code-size 8192 -o build/portisch_main_OB38S003_BUCKET_SNIFFING_INCLUDED.ihx object/portisch_main.rel object/portisch_rf_handling.rel object/portisch_serial.rel object/timer_interrupts.rel object/uart.rel object/hal.rel

?ASlink-Error-Insufficient ROM/EPROM/FLASH memory.
make: *** [Makefile:263: build/portisch_main_OB38S003_BUCKET_SNIFFING_INCLUDED.ihx] Error 1

```

I am wandering then how did you build the release from git tag v0.4.13?  Either issue with my compiler or you built from different repo state.

mightymos, 2025-02-21:
SDCC v4.0.0 is several years old. You can try using v4.5.0.

G-Tarik, 2025-02-21:
ah, those ubuntu packages. I just freshly installed it with `apt` and did not verify how old it is. I will try to get the latest one.
Thank you for the hint and replying so quickly!

G-Tarik, 2025-02-21:
It works! Thank you!

closed 2025-02-21 by G-Tarik