mightymos/RF-Bridge-OB38S003 #34: Inverted Nexus protocol in Portisch

G-Tarik, 2025-02-26:
Hello,
can someone help please to add inverted protocol in Portisch (on black box bridge). I have 2 temperature sensors from the same weather station. The protocol is Nexus, but one of the sensors sends inverted signal.

I decoded manually the "rfraw 177" sniffed  data:
```
# sensor 1
'AA B1 04 021C 0708 0366 0E42 3818181818081818181808081818181818181818181808081818181818081808080818080828 55'
# decoded bin data: 1111011110011111111110011111010001000

# sensor 2
'AA B1 04 0208 0384 074E 0E9C 3818181828182818282818181818181818282818281818182828282828181818281828181818 55'
# decoded bin data: 0001010110000000110100011111000101000
```

I added new protocol to Portisch:
```
/*
 * Nexus Temperature&Humidity Sensor
 */
#if defined(PORTISCH_SUPPORT_NEXUS_PROTOCOL)
#define NEXUS
__code static uint16_t PROTOCOL_BUCKETS(NEXUS)[] = { 500, 1000, 2000, 4000 };
__code static uint8_t PROTOCOL_START(NEXUS)[] = { LOW(3) };
__code static uint8_t PROTOCOL_BIT0(NEXUS)[]  = { HIGH(0), LOW(1) };
__code static uint8_t PROTOCOL_BIT1(NEXUS)[]  = { HIGH(0), LOW(2) };
#endif

__code static struct BUCKET_PROTOCOL_DATA PROTOCOL_DATA[] =
{
#if defined(PORTISCH_SUPPORT_NEXUS_PROTOCOL)
/*
 * Nexus
 */
  {
    { &PROTOCOL_BUCKETS(NEXUS)[0], ARRAY_LENGTH(PROTOCOL_BUCKETS(NEXUS)) },
    { &PROTOCOL_START(NEXUS)[0], ARRAY_LENGTH(PROTOCOL_START(NEXUS)) },
    { &PROTOCOL_BIT0(NEXUS)[0], ARRAY_LENGTH(PROTOCOL_BIT0(NEXUS)) },
    { &PROTOCOL_BIT1(NEXUS)[0], ARRAY_LENGTH(PROTOCOL_BIT1(NEXUS)) },
    { NULL, 0 },
    36
  },
#endif
}
```
This successfully decodes protocol on "rfraw 166" command  and sends data to mqtt but only for sensor 1. Sensor 2 never shows up with this.

I tried to change protocol like this:
```
#if defined(PORTISCH_SUPPORT_NEXUS_PROTOCOL)
#define NEXUS_INV
__code static uint16_t PROTOCOL_BUCKETS(NEXUS_INV)[] = { 500, 1000, 2000, 4000 };
__code static uint8_t PROTOCOL_START(NEXUS_INV)[] = { HIGH(3) };
__code static uint8_t PROTOCOL_BIT0(NEXUS_INV)[]  = { LOW(0), HIGH(1) };
__code static uint8_t PROTOCOL_BIT1(NEXUS_INV)[]  = { LOW(0), HIGH(2) };
#endif

```

but this time nothing is decoded.

How to add second (inverted Nexus) protocol for second sensor so both sensors work?

mightymos, 2025-02-26:
It looks like the timing buckets for sensor 2 are also different.
Buckets 1 and 2 might be swapped as compared with sensor 1.

Is that correct?

G-Tarik, 2025-02-26:
yes, that is correct.  And I think that is a sign of inverted protocol (or signal, I am not 100% sure about correct terminology).
Here is in more readable form: 
Sensor 1 buckets:
```
DEBUG <> buckets (pulse timings):
DEBUG <> bucket[0]: 0x021C == 540µs
DEBUG <> bucket[1]: 0x0708 == 1800µs
DEBUG <> bucket[2]: 0x0366 == 870µs
DEBUG <> bucket[3]: 0x0E42 == 3650µs
INFO <> bin length: 37 bit
```
Sensor 2 buckets:
```
DEBUG <> buckets (pulse timings):
DEBUG <> bucket[0]: 0x0208 == 520µs
DEBUG <> bucket[1]: 0x0384 == 900µs
DEBUG <> bucket[2]: 0x074E == 1870µs
DEBUG <> bucket[3]: 0x0E9C == 3740µs
INFO <> bin length: 37 bit
```

If my understanding of the code is correct then here
```
__code static uint16_t PROTOCOL_BUCKETS(NEXUS_INV)[] = { 500, 1000, 2000, 4000 };
```
it is just a list of buckets and the order does not have to match the order from sniffed data. Correct me please if I am wrong.

And here  by n and m we set the index (zero based) of the bucket from the buckets list
```
LOW(n), HIGH(m) 
```

mightymos, 2025-02-26:
In the data nibbles of the 0xB1 command the highest bit of each nibble indicates the logic level.
The remaining three bits are an index into the timing buckets.

So for example:
0x8 would be logic level '1' with timing bucket 0.
0x9 would be logic level '1' with timing bucket 1.

0x0 would be logic level '0' with timing bucket 0.
0x1 would be logic level '1' with timing bucket 1.

And so on.

So the sniffed 0xB1 does depend on the order of the timing buckets.
You could change the order of the timing buckets and then change the 0xB1 command to match it but I don't know why.

Ultimately what determines bit 0 or bit 1 depends on matching both logic level and timing.

To know if you are decoding correctly depends on knowing that assumptions you make about the data you are receiving is correct.
I would think both sensors should report similar temperatures if they are in the same room for example.

Are you obtaining 0xB1 command with original portisch or another tool?
Where did you get the decoded bin data from?

G-Tarik, 2025-02-26:
I started with original Portisch a month or so ago, the rf bridge was constantly just in rfraw 177 sniffing  mode and decoded data perfectly matches temperature, humidity values, and devices IDs are stable so I always know which device sent data.  Then I compiled your fork with sniffing enabled and everything work exactly the same.  So now I just decided to add protocol definition to not run it in constant sniffing mode.

For decoding 0xB1 data I followed original Portisch wiki and additional help from some articles and I created my own Python script. So sniffed data is decoded in mqtt subscriber and is sent to influxdb and this is reliable. Later on I discovered the PiLight library where I found that my protocol actually called Nexus  https://github.com/pilight/pilight/blob/staging/libs/pilight/protocols/433.92/nexus.c

The key part of my decoding Python script is here which includes inversion detection:
```Python
sniff = '3818181818081818181808081818181818181818181808081818181818081808080818080828'
raw = sniff[1:-1]
bin_data = [int(int(raw[i:i + 2], 16) & 0x77 == 1) ^ inverse for i in range(0, len(raw), 2)]
```
where `inverse` is either `False` for sensor 1  or `True` for sensor 2.

G-Tarik, 2025-02-26:
IDs change if battery changed. Here is decoded binary data into human values:
```
s1 bin: 1111011110011111111110011111010001000 <> t: -0.7 h: 68 id: 247 <> bat:1 ch:1
s2 bin: 0001010110000000110100011111000101000 <> t: 20.9 h: 20 id: 21 <> bat:1 ch: 0
```

mightymos, 2025-02-26:
For sensor 1, bit0 = 0x80 and bit1 = 0x81
For sensor 2, bit0 = 0x81 and bit1 = 0x82

In both sensors and both bits the logical high level is followed by a logical low level according to 0xB1 output.
In other words, the first nibble is always highest bit set (i.e., 0x8), followed by three bits representing which bucket timing.
For the second nibble the highest bit is always cleared, followed by the bucket timing.

For sensor 1 the bucket times are bucket0 = 0x021C (540), bucket 1 = 0x0708 (1800)
For sensor 2 the bucket times are bucket0 = 0x0208 (520), bucket 1 = 0x0384 (900), and bucket 2 = 0x074E (1870)

So for sensor 2 bit 0 is HIGH(0) LOW(1) and bit 1 is HIGH(0) LOW(2)

On both sensors the start bit seems to be 0x3 or LOW(3) and end bit seems to be 0x8 or HIGH(0)

However, I am confused by the 0x82 0x8 at the end of the sensor 1 decoding.
I will ignore that issue for now.

Can you try these mappings for sensor 2?

G-Tarik, 2025-02-27:
Jonathan, I appreciate a lot your help, thank you!

It almost works, apparently I will need just to add some post-processing for sensor 2.

So with this protocol definition:
```cpp
#if defined(PORTISCH_SUPPORT_NEXUS_PROTOCOL)
#define NEXUS
__code static uint16_t PROTOCOL_BUCKETS(NEXUS)[] = { 500, 1000, 2000, 4000 };
__code static uint8_t PROTOCOL_START(NEXUS)[] = { LOW(3) };
__code static uint8_t PROTOCOL_BIT0(NEXUS)[]  = { HIGH(0), LOW(1) };
__code static uint8_t PROTOCOL_BIT1(NEXUS)[]  = { HIGH(0), LOW(2) };
//__code static uint8_t PROTOCOL_END(NEXUS)[]   = { HIGH(0) };  // with or without the result is the sam
e
......

#endif
#if defined(PORTISCH_SUPPORT_NEXUS_PROTOCOL)
>·>·/*
>·>· * Nexus
>·>· */
>·>·{
>·>·>·{ &PROTOCOL_BUCKETS(NEXUS)[0], ARRAY_LENGTH(PROTOCOL_BUCKETS(NEXUS)) },
>·>·>·{ &PROTOCOL_START(NEXUS)[0], ARRAY_LENGTH(PROTOCOL_START(NEXUS)) },
>·>·>·{ &PROTOCOL_BIT0(NEXUS)[0], ARRAY_LENGTH(PROTOCOL_BIT0(NEXUS)) },
>·>·>·{ &PROTOCOL_BIT1(NEXUS)[0], ARRAY_LENGTH(PROTOCOL_BIT1(NEXUS)) },
>·>·>·{ NULL, 0 },
>·>·>·36
>·>·},
#endif

```

all sensors decoded properly except sensor 2.
I have more than 2 sensors actually, of different models but the same protocol.
As you pointed out the 0xB1 produces the size of 37 bits, I was also just ignoring that while decoding worked. And despite 36 bit in protocol definition the 0xA6 result also produces 37 bit.

Now the problem is with sensor 2:

sniff 0xB1 result decoded by my Python script:
t: 20.9 h: 20 (from display);  expected id: 21
1110101001111111001011100000111010111 - I see that values make no sense so I set `inverse` flag to True
0001010110000000110100011111000101000 - now values match and id is as expected.

now 0xA6 result:
t: 19.3C h: 30% (from display); expected id: 21
rfraw: AAA606001580C1F1E055
bin(0x1580C1F1E0): 1010110000000110000011111000111100000 - after trying to decode the values make no sense, id is unexpected

but if I prepend leading zeros then it perfectly matches the format what is produced by 0xB1:
1010110000000110000011111000111100000 - received 0xA6
0001010110000000110000011111000111100 - transformed, now values make sense

So apparently 0xA6 is stripping leading zeros and the start of the payload is the first logical "1" and in the end it appends zeros to match the protocol size.

If you have some ideas how to fix this in the code then it would be nice, but if not then it is not critical, I can deal with this on mqtt subscriber side.

A side discussion: I wanted to make 0xA6 mode work mainly just out of curiosity and having fun challenge. Although without your effort of porting Portisch to sdcc I would not be able to do anything. But practically from my perspective all the hassle with adding the protocol and compilation and still need of decoding the result of 0xA6 is not worth it compared just to leave 0xB1 mode running and anyway do the decoding by yourself. I mean the user still needs to now the exact protocol to parse real values out of binary result from 0xA6. So from user perspective there is no that big difference if one parses 0xA6, or if one parses 0xB1  - just one Python line more. The only noticeable difference is a bit more noise coming to mqtt broker from 0xB1 sniffing but it can be easily just ignored by mqtt subscriber.  
What is your opinion on using 0xA6 with added custom protocol versus using 0xB1 mode?

mightymos, 2025-02-27:
I noticed this note in the nexus source:
```
// the first and second-to-last pulse should be a short one
```

I have been trying to understand the nexus decoding source files.
I think you should try omitting the start bit (make array null and size 0).
And then have the end bit equal to the low level for the sync time (4000 microsec).

What I think might happen is portisch will detect the gap between repeat transmission as the sync.
Then hopefully correctly decode the data bits.

G-Tarik, 2025-02-28:
good point about `// the first and second-to-last pulse should be a short one` !

but in Portisch it does not change anything, with these variants the result is same as above, sensor 2 still has the same anomaly
```cpp
__code static uint16_t PROTOCOL_BUCKETS(NEXUS)[] = { 500, 1000, 2000, 4000 };
// op1
__code static uint8_t PROTOCOL_BIT0(NEXUS)[]  = { HIGH(0), LOW(1) };
__code static uint8_t PROTOCOL_BIT1(NEXUS)[]  = { HIGH(0), LOW(2) };
__code static uint8_t PROTOCOL_END(NEXUS)[]   = { LOW(3) };

/ opt2
__code static uint8_t PROTOCOL_BIT0(NEXUS)[]  = { HIGH(0), LOW(1) };
__code static uint8_t PROTOCOL_BIT1(NEXUS)[]  = { HIGH(0), LOW(2) };
__code static uint8_t PROTOCOL_END(NEXUS)[]   = { HIGH(0), LOW(3) };

// opt3
__code static uint8_t PROTOCOL_START(NEXUS)[] = { LOW(3) };
__code static uint8_t PROTOCOL_BIT0(NEXUS)[]  = { HIGH(0), LOW(1) };
__code static uint8_t PROTOCOL_BIT1(NEXUS)[]  = { HIGH(0), LOW(2) };
__code static uint8_t PROTOCOL_END(NEXUS)[]   = { HIGH(0), LOW(3) };

// opt4
__code static uint8_t PROTOCOL_START(NEXUS)[] = { LOW(3) };
__code static uint8_t PROTOCOL_BIT0(NEXUS)[]  = { HIGH(0), LOW(1) };
__code static uint8_t PROTOCOL_BIT1(NEXUS)[]  = { HIGH(0), LOW(2) };
__code static uint8_t PROTOCOL_END(NEXUS)[]   = { LOW(3) };
```

with this variant there is nothing at all, radio silence:
```
__code static uint16_t PROTOCOL_BUCKETS(NEXUS)[] = { 500, 1000, 2000, 4000 };
__code static uint8_t PROTOCOL_START(NEXUS)[] = { HIGH(0) };
__code static uint8_t PROTOCOL_BIT0(NEXUS)[]  = { HIGH(0), LOW(1) };
__code static uint8_t PROTOCOL_BIT1(NEXUS)[]  = { HIGH(0), LOW(2) };
__code static uint8_t PROTOCOL_END(NEXUS)[]   = { HIGH(0), LOW(3) };
```

Very interesting fact the I am running MQTT433gateway (just with updated ESPiLight/pilight libs) on nodemcu v3 with RXB6 receiver and it decodes (to the real temp/humidity values) all my sensors out of the box without any protocol adjustments, even sensor 2.

Yes, Zegbee might be better choice for the new fresh setup. I just had already these sensors in place so just utilizing resources.

So I don't want to waste your time, the current result even with additional scripting is acceptable for me. We can close the issue. 
But of course I am available for some tests if you have any ideas.
Thank you for your help!

mightymos, 2025-02-28:
Okay, good that it is working in some capacity.
If we had an oscilloscope or raw signal capture we could know definitely.
But that work may be unnecessary when other working decoders are available.

Thank you for mentioning the openmqtt and related projects.
I will try to remember to list them in the readme as alternative options.

Could you please just take a look at this other users MQTT stuff and comment if you can help them?
https://github.com/mightymos/OnbrightFlasher/issues/10

They might have already figured things out for themselves but I don't know MQTT.

closed 2025-02-28 by mightymos