mightymos/RF-Bridge-OB38S003 #36: Re-sending RfRaw codes sniffed on old Sonoff (black box) with Tasmota

sstefanowski, 2025-04-01:
Hello,
My old Sonoff with Porthish (balck Sonoff) has died and it was my motivation to try this software.

After a few hours of fighting, I was able to flash  MCU with [portisch_main_OB38S003_MULTIPLE_PROTOCOLS_INCLUDED.hex](https://github.com/mightymos/RF-Bridge-OB38S003/releases/download/v0.4.16/portisch_main_OB38S003_MULTIPLE_PROTOCOLS_INCLUDED.hex) image and basic tasmota-sensors 14.5.0, set Module to `Generic (0)`, set TX to `RFSend`, RX to `RFrecv` and D7 GPIO13 to `LedLink`

Now, I have couple of old codes, which were hijacked on black-Sonoff and converted from B1 to B0 with BitBusket Coverter.
I try to sending command on new white Sonnof on the console but nothing happens - error `Command unknown` 

```
19:28:14.111 CMD: RfRaw AA B0 4B 04 08 0276 0140 00D2 100E B081A081A0A0A08181A0A0A0A0A0A0A0A0A0A0A090A0A0A0A090818181A08181A081A08181818181A0A0A0A0A0A0A0A090A090A09081908181A0A0818181818181 55
19:28:14.124 RSL: RESULT = {"Command":"Unknown","Input":"RFRAW AA B0 4B 04 08 0276 0140 00D2 100E B081A081A0A0A08181A0A0"}
```
How to send old RF codes on Tasmota?

mightymos, 2025-04-01:
You need to set the module to Sonoff (25) since you're sending serial bytes.
The Generic (0) is only if you're directly manipulating the pins (i.e. passthrough mode).

sstefanowski, 2025-04-01:
Hmmm.... After changing to SonofBridge(25) it's  still no response. The console log shows:

mightymos, 2025-04-01:
Ok, you are getting an acknowledge back from the microcontroller so it is at least "alive" and should actually be transmitting.
The only thing I can suggest is to flash the SNIFFING firmware and try to resniff the code for your device.
That way we could at least compare or see that something is working.

What kind of device are you controlling (e.g., light, blinds, etc.)?
Do you have the original remote?

sstefanowski, 2025-04-01:
Well... OK I understand...

I'd like to control the screen for the projector (so it;'s simple controller). But black-sonoff was working fine for several years and... I need to search for the original remote... 

Sorry for noob questions.... So I need to 
1) re-flash MCU with SNIFFING firmware 
2) issue "rfraw AAB155"?? and start pressing remote buttons?
3) re-flash MCU with MULTIPLE_PRTOCOL firmware and play B0 code (produced from sniffed B1)

Am I right or I miss something?

Do I need to do step 3?

mightymos, 2025-04-01:
I am not sure if it will help to resniff but it might.
You do not need to do step 3, the sniffing mode firmware can also replay B0 codes.

sstefanowski, 2025-04-01:
I managed to do it quickly. Reflash MCU without erasing Tasmota. 
Got the original remote (luckily, I had new batteries :) ), sniffed B1 codes, and then generated B0, and... transmission works! 
Now I'll test the stability of the receiving mode as I have a few RF433 buttons in various places of my home to control lights. In case receiving hangs, I'll add automation to reset Sonoff overnight.
It's really great work!

mightymos, 2025-04-01:
Great.

It would be useful to me to compare the measured codes/timings?
Can you post some of your old B0 codes along with resniffed B0 codes?

sstefanowski, 2025-04-01:
Sure, here you are....

Old DOWN
AA B0 4B 04 08 0276 0140 00D2 100E B081A081A0A0A08181A0A0A0A0A0A0A0A0A0A0A090A0A0A0A090818181A08181A081A08181818181A0A0A0A0A0A0A0A090A090A09081908181A0A0818181818181 55

Old UP
AA B0 4B 04 08 026C 0140 00D2 102C B081A081A0A0A08181A0A0A0A0A0A0A0A0A0A0A0A0A0A0A0A0A0818181A08181A081A08181818181A0A0A0A0A0A0A090A0A081A0A0A0908181A0818181A0818181 55

New DOWN
AA B0 4B 04 08 026E 014A 00C8 0FE1 B081A081A0A0A08181A0A0A0A0A0A0A0A0A0A0A0A0A0A0A0A0A0818181A08181A081A08181818181A0A0A0A0A0A0A0A0A0A0A0A0A081A08181A0A0818181818181 55

New UP
AA B0 4B 04 08 026B 0144 00C9 0FE3 B081A081A0A0A08181A0A0A0A0A0A0A0A0A0A0A0A0A0A0A0A0A0818181A08181A081A08181818181A0A0A0A0A0A0A0A0A0A081A0A0A0A08181A0818181A0818181 55

mightymos, 2026-08-16:
@sstefanowski 

If you're willing to provide some feedback or measurements using the new firmware that could be helpful:
https://github.com/mightymos/RF-Bridge-OB38S003/releases/tag/v0.9.0

However, if you're setup is working and you don't want to mess with it I understand.