mightymos/RF-Bridge-OB38S003 #37: unable to flash firmware with tasmota loaded - unable to flash ESP with firmware flashed

ale624, 2025-04-16:
Hi,

I am having an issue where i cannot flash my RF firmware with tasmota installed. but if i erase the ESP i am unable to flash the ESP after RF firmware is flashed. if i erase the RF chip i can flash the ESP.

i am using a Sonof R2 2.2.
are there some traces i can cut and then resolder to make this doable?? 
i'm at my whits end!

error i get during RF flashing is: 
2025-04-17 00:30:31,970 - ERROR - Sent line representation: ':0300000002004CAF'
2025-04-17 00:30:31,970 - ERROR - Response representation: 'Status: 4'
2025-04-17 00:30:31,970 - ERROR - State 8 failed. Returning to handshake state.

i do not mind performing HW mods to make this work. 
i can get the passthru FW to flash with the ESP wiped. the portish firmware failes with similar error to above.

mightymos, 2025-04-17:
Because flashing the the OB38S003 microcontroller uses the I2C bus without any other devices connected it should always be possible to flash. This assumes a few things like a sufficient power supply.

However, passthrough firmware in particular can interfere with flashing the ESP8285 because it manipulates the same pins while the PC is trying to communicate over USB-Serial.

There should not be any difference other than that with flashing passthrough versus portisch.
Which one do you need, if you do not know your devices you should probably use the sniffing mode with portisch to start?

ale624, 2025-04-17:
I've adjusted how i am powering the device, but i am still unable to get any other firmware to flash. i've soldered wires rather than using jumper cables to ensure the connection is as stable as possible. i can only get the passthrough to flash. the other firmwares fail half way through their flashing

are you able to add some kind of retry or delay mechanism to the writing proccess??

mightymos, 2025-04-17:
I did not write the flashing script so I am hesitant to try changing it at the moment.
What board are you using as a flasher?

ale624, 2025-04-19:
An ESP8366

mightymos, 2025-04-19:
It's just very unusual.
There's no need to change it since you already soldered but I would normally recommend installing headers and then using premade jumper wires as needed.

Can you attach a picture of the flashing setup?