mightymos/RF-Bridge-OB38S003 #38: Problems Sending Commands with Sonoff RF Bridge R2 v2.2

markuspinter, 2025-05-03:
Hi, I’m trying to control my shutters using a Sonoff RF Bridge R2 v2.2.

I flashed tasmota and the portisch multiple protocols hex file.

When trying to sniff the commands from my remote I only get responses using
rfraw 166
While rfraw 177 don’t yield any results. 

However, attempting to send those received commands doesn’t trigger the shutters in any way.
Here’s my console output:

```
12:11:08.902 CMD: rfraw 166
12:11:08.907 RSL: RESULT = {"RfRaw":"ON"}
12:11:08.928 RSL: RESULT = {"Time":"2025-05-03T12:11:08","RfRaw":{"Data":"AAA055"}}
12:11:10.579 RSL: RESULT = {"Time":"2025-05-03T12:11:10","RfRaw":{"Data":"AAA6060583468B443355"}}
12:11:10.758 RSL: RESULT = {"Time":"2025-05-03T12:11:10","RfRaw":{"Data":"AAA6060583468B443C55"}}
12:11:12.161 RSL: RESULT = {"Time":"2025-05-03T12:11:12","RfRaw":{"Data":"AAA6060583468B445555"}}
12:11:12.836 RSL: RESULT = {"Time":"2025-05-03T12:11:12","RfRaw":{"Data":"AAA6060583468B443355"}}
12:11:13.675 RSL: RESULT = {"Time":"2025-05-03T12:11:13","RfRaw":{"Data":"AAA6060583468B443355"}}
12:11:14.515 RSL: RESULT = {"Time":"2025-05-03T12:11:14","RfRaw":{"Data":"AAA6060583468B443355"}}
12:11:15.351 RSL: RESULT = {"Time":"2025-05-03T12:11:15","RfRaw":{"Data":"AAA6060583468B443355"}}
12:11:16.531 RSL: RESULT = {"Time":"2025-05-03T12:11:16","RfRaw":{"Data":"AAA6060583468B445555"}}
12:11:26.780 CMD: rfraw 0
12:11:26.785 RSL: RESULT = {"RfRaw":"OFF"}
12:12:07.293 CMD: rfraw AAA8550583468B443355
12:12:07.299 RSL: RESULT = {"RfRaw":"ON"}
12:12:19.209 CMD: rfraw 0
12:12:19.213 RSL: RESULT = {"RfRaw":"OFF"}
12:14:50.851 CMD: rfraw AAA855
12:14:50.856 RSL: RESULT = {"RfRaw":"ON"}
12:15:26.473 CMD: rfraw 0583468B443355
12:15:26.478 RSL: RESULT = {"RfRaw":"ON"}
12:15:30.572 CMD: rfraw 0583468B443355
12:15:30.577 RSL: RESULT = {"RfRaw":"ON"}
```

Sending rfraw AAA8550583468B443355 doesn’t do anything either, sending the other commands also don’t trigger anything.

Could you help me, please? I’m not sure what I’m doing wrong, and I’m having a hard time understanding the portisch docs.

Thanks so much!

Best
Markus

mightymos, 2025-05-04:
You would need to flash the sniffing firmware to use 0xB1 command.

markuspinter, 2025-05-04:
Okay thank you

mightymos, 2025-05-05:
Let me know if you have any progress?