mightymos/RF-Bridge-OB38S003 #39: Difference between SNIFFING INCLUDED and PROTOCOLS INLCUDED

JMDesigns123, 2025-05-17:
I use the the version 0.4.12 from main_portisch_OB38S003.hex. What is the difference of the version 0.4.16 between portisch_main_OB38S003_BUCKET_SNIFFING_INCLUDED.hex and portisch_main_OB38S003_MULTIPLE_PROTOCOLS_INCLUDED.hex?

Thanks for the feedback!

mightymos, 2025-05-17:
Portisch allows sniffing (and replay) of unknown protocols.
It does this using some basic rules to start such as looking for a gap between repeat transmissions.
And then forming a table of measured bucket timings to report after at least two repeats are detected.

The multiple protocols has a look up table which measured timings are compared with to look for a a match.

The reason for having two different firmwares is there is not enough code space and ram to have both features included at the same time.

pawol, 2026-01-14:
Which one is suggested to use?

mightymos, 2026-01-14:
> Which one is suggested to use?

Portisch is used when you have no idea what timing protocol your device uses.
Protocols Included only has a few protocols included (but there are hundreds in use in the world).