mightymos/RF-Bridge-OB38S003 #42: logging confuses timing that happens to contain the end sentinal value with a payload terminator

axa88, 2025-06-29:
Below you can see the same command sent 3 times, the last time the timing happens to contain '55', seems logging gets this confused with a termination sentinel and handles it incorrectly.

You can see the uart contains the correct data but perhaps this is a data point to use in finding incorrect parsing and instability in the firmware.

This was caught while enabling portisch bucket sniffing.

```
[02:37:22][I][rf_bridge:104]: Received RFBridge Bucket: AA B1 03 04AE 017A 2E51 28190909090819081908190819081819090909090909090819 55
[02:37:22][D][text_sensor:064]: 'Rx RF': Sending state 'aab10304ae017a2e512819090909081908190819081908181909090909090909081955'

[02:37:23][I][rf_bridge:104]: Received RFBridge Bucket: AA B1 03 04AE 0179 2E51 28190909090819081908190819081819090909090909090819 55
[02:37:23][D][text_sensor:064]: 'Rx RF': Sending state 'aab10304ae01792e512819090909081908190819081908181909090909090909081955'

[02:37:24][I][rf_bridge:104]: Received RFBridge Bucket: AA B1 03 04AD 0179 2E 55 
[02:37:24][D][text_sensor:064]: 'Rx RF': Sending state 'aab10304ad01792e552819090909081908190819081908181909090909090909081955'
```

mightymos, 2025-06-29:
Yeah, unfortunately the sniffing command does not contain a length field. I do not recall if that is because it sends out as it is decoding or not. But some of the other commands have length bytes that could be used to know what position to expect a true payload end byte.

axa88, 2025-06-30:
> Yeah, unfortunately the sniffing command does not contain a length field.

I see.  Though in this instance you know from data this is not end sentinel but rather it is timing data, no? You might wait a certain amount of time to see next byte is not start sentinel? 

Well as long as print log does not corrupt uart , perhaps it is non critical issue. I wouldn't know. 

I just wish to help improve stability

axa88, 2025-07-05:
meh , its just the log, focus on other things..… closed

closed 2025-07-05 by axa88