mightymos/RF-Bridge-OB38S003 #43: Failing to aquire data from

axa88, 2025-07-02:
Using the 

im using the `portisch_main_OB38S003_BUCKET_SNIFFING_INCLUDED.hex` fw along with esphome and i am not getting any data from `rf_bridge: on_advanced_code_received` .  This after triggering `rf_bridge.start_advanced_sniffing` for good measure.

Im just not sure what i might be doing wrong, or if its even available for this particular fw.

Please advise

axa88, 2025-07-02:
> Does standard decoding provide any output either?

standard decoding? meaning the value `on_code_received` returns?
yes that returns a value.  just trying to understand everything i can...

mightymos, 2025-07-02:
Yes that's right.
That mode basically uses rules to discern what is considered similar to a standard timing.
But advanced compares to a look up table.

closed 2025-07-05 by axa88