mightymos/RF-Bridge-OB38S003 #44: Unable flashing RF chip

RecepAGulec, 2025-09-18:
Hello,
I deleted the ESP chip and RF chip as instructed on the "Flashing guide by example" page.
Now it's time to "Flashing the custom firmware to the MCU (radio chip)." I ran flashScript.py, selected the COM port and the "passthrough_main_OB38S003.hex" file, and connected 3.3V to the RF bridge circuit.
While hex file has been writing, I received the following error message:
2025-09-18 21:34:44,801 - INFO - Wrote 16 bytes
2025-09-18 21:34:44,808 - INFO - :10002800E0B50202EBF02230F6E0A88220F5D3EA30
2025-09-18 21:34:44,826 - INFO - Write successful
2025-09-18 21:34:44,827 - INFO - Wrote 16 bytes
2025-09-18 21:34:44,834 - INFO - :10003800C6F582228B822230F6E6A88220F5D98086
2025-09-18 21:34:44,848 - INFO - Status: 2
2025-09-18 21:34:44,849 - ERROR - Error: Unexpected response or mismatch with sent line.
2025-09-18 21:34:44,849 - ERROR - Sent line representation: ':10003800C6F582228B822230F6E6A88220F5D98086'
2025-09-18 21:34:44,850 - ERROR - Response representation: 'Status: 2'
2025-09-18 21:34:44,851 - ERROR - State 8 failed. Returning to handshake state.
2025-09-18 21:34:44,852 - INFO - NACK on transmit of address
2025-09-18 21:34:44,852 - INFO - Status: 2
I tried the operation many times, but I always got the same error message.
Where am I making a mistake? My RF Bridge is R2 v2.2.

mightymos, 2025-09-18:
There have also been some issues with certain versions of the Arduino with ESP32:
https://github.com/mightymos/RF-Bridge-OB38S003/issues/40

RecepAGulec, 2025-09-29:
I don't think the problem is with the programming hardware. In my experience, the problem is with the hex file. I tried to install the files "passthrough_main_OB38S003.hex" and "portisch_main_OB38S003_MULTIPLE_PROTOCOLS_INCLUDED.hex" several times without success. However, the file "RF-Bridge-OB38S003_SimplePassThrough_01-14-2024.hex" installed on the first try.